hiyohiyo/CrystalDiskInfo #15: asi.HostReads

adler3d, 2018-08-07:
https://github.com/hiyohiyo/CrystalDiskInfo/blob/ffc641d53893ebd91d73324b790d2d565afd6207/AtaSmart.cpp#L3091

how about:
asi.HostReads = (*(ULONG64*)&asi.SmartReadData[0x20]) * 512 / 1024 / 1024;

adler3d, 2018-08-09:
also look at this:
https://github.com/mirror/smartmontools/blob/fac3b72ddc5224cbe490544844bc326ed50e5246/nvmecmds.h#L187

adler3d, 2018-08-09:
also "512 / 1024 / 1024" is wrong.
use "(512*1000)/(1024*1024*1024)" instead.

hiyohiyo, 2020-06-28:
It was fiexed...
----
		asi.HostReads = (ULONG64)
		         (((ULONG64)asi.SmartReadData[0x27] << 56)
				+ ((ULONG64)asi.SmartReadData[0x26] << 48)
				+ ((ULONG64)asi.SmartReadData[0x25] << 40)
				+ ((ULONG64)asi.SmartReadData[0x24] << 32)
				+ ((ULONG64)asi.SmartReadData[0x23] << 24)
				+ ((ULONG64)asi.SmartReadData[0x22] << 16)
				+ ((ULONG64)asi.SmartReadData[0x21] << 8)
				+ ((ULONG64)asi.SmartReadData[0x20])) * 512 * 1000 / 1024 / 1024 / 1024;

closed 2020-06-28 by hiyohiyo