hiyohiyo/CrystalDiskInfo #16: Advanced format 4Kn sector device identification is incorrect

nasbdh9, 2018-09-10:
[image]
This is a 4Kn sector device but the capacity identification is not correct, maybe this is because of windows api problem?

Is it possible to record all Smart? Currently, recording only the current value may be inaccurate for judging certain hard drives.

nasbdh9, 2020-06-21:
long time ago, I realized that this may be a Windows API problem. Under the native 4Kn format hard drive, the correct capacity should be eight times the displayed value

kravena, 2020-07-22:
I am experiencing this same issue after changing from 512e to 4kn

hiyohiyo, 2020-07-22:
@kravena I would like you to send text copy to me. hiyohiyo@crystalmark.info or this thread. I will support near future.

kravena, 2020-07-23:
I can't give you full info, as I have the drives raided, but I have provided what I can below :

However, these are 14TB WD Drives WD140EMFZ which I have reformatted from 512e to 4kn using the WD Hugo tool. I also did the same with the 12TB WD Gold drives I have WD121KRYZ which has the same issue in Diskinfo.

[
[drive.txt](https://github.com/hiyohiyo/CrystalDiskInfo/files/4965189/drive.txt)
](url)

[diskinfo.txt](https://github.com/hiyohiyo/CrystalDiskInfo/files/4965214/diskinfo.txt)
[image]

Hugo command used : 
 f -m <DRIVE MODEL> --fastformat -b 4096

hiyohiyo, 2020-07-23:
Thank you detail information. I will support 4Kn future release!!

hiyohiyo, 2020-07-24:
CrystalDiskInfo 8.8.0 Alpha1 or later supports 4Kn. Please test it.
https://crystalmark.info/en/

nasbdh9, 2020-07-24:
[image]

Tested in 8.8.0 Alpha1, shows the correct capacity👍

kravena, 2020-07-25:
Yes, it works

hiyohiyo, 2020-07-25:
CrystalDiskInfo 8.8 will be released near future.

closed 2020-07-25 by hiyohiyo