hiyohiyo/CrystalDiskInfo #18: intel s3500 (ver DELL EMC OEM) support?

kumoilain, 2019-01-03:
Will you support this SSD in future? Or can SMART information be modified by users?
Thanks!
[image]
[image]
Port SMART info from intel ssd toolbox:
[s3500.txt](https://github.com/hiyohiyo/CrystalDiskInfo/files/2724398/s3500.txt)

kumoilain, 2019-02-05:
[image]
[image]
[image]
[image]
These screenshots are taken from Intel SSD Toolbox. Thanks for your work.

hiyohiyo, 2020-08-11:
CrystalDiskInfo 8.8.0 Alpha7
- Added Intel SSD DC 3500 Series

closed 2020-08-11 by hiyohiyo