hiyohiyo/CrystalDiskInfo #21: WD blue 3D NAND SSD support

hiyohiyo, 2019-02-05:
[image]

krrr, 2019-05-20:
It seems that WD blue 3d and SanDisk ultra 3d are the same

http://www.thessdreview.com/our-reviews/wd-blue-3d-ssd-sandisk-ultra-3d-ssd-review-1tb-twins-at-their-finest/2/
[image]
[image]

hiyohiyo, 2020-08-02:
CrystalDiskInfo 8.8.0 Alpha2 supports WD/SanDisk SATA SSDs.

closed 2020-08-02 by hiyohiyo