hiyohiyo/CrystalDiskInfo #54: Improve Send Mail feature

hiyohiyo, 2020-07-24:
https://osdn.net/projects/crystaldiskinfo/ticket/32029

hiyohiyo, 2020-07-24:
Custom defined description support.
https://osdn.net/projects/crystaldiskinfo/ticket/28716

メール本文は柔軟にカスタマイズできるようにしたいですね・・・。

hiyohiyo, 2020-07-24:
.NET Failure Message with SMTP invalid certificate
https://osdn.net/projects/crystaldiskinfo/ticket/36624

エラーが出て動かないとのこと。

yatounoneko, 2022-02-03:
Sorry, I found this issues is still persist.
[image]

Someone also said that Gmail could not be used on reddit. Looks like the TLS version support problem.
[](https://www.reddit.com/r/DataHoarder/comments/f0tlla/crystaldiskinfo_gmail_email_notification_the/)