hiyohiyo/CrystalDiskInfo #81: Latest AMD StoreMI causes "disk not found"

Josephur, 2020-08-18:
Recently built me a new beast of a machine, and installed the latest StoreMI (it's been re-vamped and re-released recently, does not even work how the old one did)

This caused all of my disks to be recognized as SCSI devices (even though they aren't) to add support for StoreMI's driver.

Also added was "StoreMI Controller" and "StoreMI Bottom Device" in device manager. 

But now CrystalDiskInfo sees no hard drives at all :(

I'm not sure if there is a work around, but I didn't see a similar issue posted, so I thought it was worth posting about since many users use StoreMI.

hiyohiyo, 2020-08-24:
Unfortunately, CrystalDiskInfo does not support StoreMI 1.0/2.0 currently.

mirko8054, 2023-03-27:
Bumping this, hopefully one day we will see StoreMi support on DiskInfo