hiyohiyo/CrystalDiskInfo #82: m.2 drive and pcie drive are mixed in one entity

Albirew, 2020-08-20:
I have a crucial ct250 m.2 250GB ssd used for system (c:), a Sandisk sx350 pcie 3200GB ssd for data (e:) and 2 hdd (b: & d:)
both ssd data are mixed together (see scrennshot), especially visible with the tooltip (250gb in name and 2900GB with e: letter below).

From what i remember, firmware and startup times are from 250gb disk,, interface and transfer mode are out from a hat (m.2 and pcie, not sata) and powered-on time and host write total are from pcie card.

[image]

hiyohiyo, 2020-08-24:
Current drive letter detection feature is not perfect.
I would like to improve it in the future release.

Albirew, 2025-08-17:
i'm currently cleaning my old issues that are still open.
since last activity is from 2020, should i close this issue?
(on 9.7.1, my 3Tb PCIe SSD doesn't appear anymore)