hiyohiyo/CrystalDiskInfo #86: How w to NVME interface hard disk master and firmware information?

lwei2, 2020-09-21:
I want to get the master control and firmware information of the NVME interface hard disk. I read and write by IOCTL_SCSI_PASS_THROUGH control code, but the result returns nothing.Please give me a hand.

hiyohiyo, 2020-09-21:
NVME interface hard disk???

Please read an AtaSmart.cpp.

lwei2, 2020-09-22:
> NVME接口硬盤？？？
> 
> 請閱讀AtaSmart.cpp。

I have seen the AtaSmart.cpp source file,and  an attempt to get the master and firmware information of the NVME interface returns null in the structure of the device identifier(GetLastError()==0). Do not know how to get its master control and firmware information?Could you help me, please?

hiyohiyo, 2020-09-22:
Do you use Windows 10?

DoIdentifyDeviceNVMeStorageQuery is only support on Windows 10.

lwei2, 2020-09-22:
> Do you use Windows 10?
> 
> DoIdentifyDeviceNVMeStorageQuery is only support on Windows 10.

Yes,I use Windows 10,and I try to use the DoIdentifyDeviceNVMeStorageQuery  interface to get master and firmware information ,it returns an empty.The DoIdentifyDeviceNVMeIntel interface, DoIdentifyDeviceNVMeJMicron interface , DoIdentifyDeviceNVMeASMedia interface , DoIdentifyDeviceNVMeRealtek interface  also get empty.

hiyohiyo, 2020-09-22:
Sorry, I have no idea.
My knowledge puts CrystalDiskInfo source code...

Please test Hard Disk Sentinel.
https://www.hdsentinel.com/

lwei2, 2020-09-22:
> Sorry, I have no idea.
> My knowledge puts CrystalDiskInfo source code...
> 
> Please test Hard Disk Sentinel.
> https://www.hdsentinel.com/

Thank you.

lwei2, 2020-09-22:
> Sorry, I have no idea.
> My knowledge puts CrystalDiskInfo source code...
> 
> Please test Hard Disk Sentinel.
> https://www.hdsentinel.com/

>

lwei2, 2020-09-22:
> > Sorry, I have no idea.
> > My knowledge puts CrystalDiskInfo source code...
> > Please test Hard Disk Sentinel.
> > https://www.hdsentinel.com/
> 
> >

hi, hiyohiyo.Excuse me, I have another question to ask you.The CrystalDiskInfo software can obtain the UASP(NVM Express)  interface of the hard disk master and firmware information.At present, there are different firmware, interfaces are UASP (NVM Express) two pieces of hard disk, the call DoIdentifyDeviceNVMeSamsung after this interface, one can read out the firmware information, another is read out information is empty.Therefore, I would like to ask if there is any way to obtain the master and firmware information of the UASP(NVM Express) interface hard disk.

hiyohiyo, 2020-09-22:
Does ”hard disk master” mean Model Name?

DoIdentifyDeviceNVMeSamsung is using Samsung specific interface.

- DoIdentifyDeviceNVMeJMicron
- DoIdentifyDeviceNVMeASMedia
- DoIdentifyDeviceNVMeRealtek

are using vendors specific interface.

lwei2, 2020-09-22:
> Does ”hard disk master” mean Model Name?
> 
> DoIdentifyDeviceNVMeSamsung is using Samsung specific interface.
> 
> * DoIdentifyDeviceNVMeJMicron
> * DoIdentifyDeviceNVMeASMedia
> * DoIdentifyDeviceNVMeRealtek
> 
> are using vendors specific interface.

”hard disk master”  mean master control of solid state disk.

lwei2, 2020-09-22:
> Does ”hard disk master” mean Model Name?
> 
> DoIdentifyDeviceNVMeSamsung is using Samsung specific interface.
> 
> * DoIdentifyDeviceNVMeJMicron
> * DoIdentifyDeviceNVMeASMedia
> * DoIdentifyDeviceNVMeRealtek
> 
> are using vendors specific interface.
I find 

> > Does ”hard disk master” mean Model Name?
> > DoIdentifyDeviceNVMeSamsung is using Samsung specific interface.
> > 
> > * DoIdentifyDeviceNVMeJMicron
> > * DoIdentifyDeviceNVMeASMedia
> > * DoIdentifyDeviceNVMeRealtek
> > 
> > are using vendors specific interface.
> 
> ”hard disk master” mean master control of solid state disk.

Sorry to bother you again, Hiyohiyo. I find the software of CrystalDiskInfo 8.8.7 does not read the hard disk of UASP (NVM Express) interface, but the software of CrystalDiskInfo 8.7.0 does read the hard disk of the two UASP (NVM Express). Where can the source code of CrystalDiskInfo 8.7.0 be downloaded? I want to debug through VS2017 and locate how the code gets its firmware information. Could you please tell me where CrystalDiskInfo 8.7.0 source code can be downloaded?

lwei2, 2020-09-22:
> Does ”hard disk master” mean Model Name?
> 
> DoIdentifyDeviceNVMeSamsung is using Samsung specific interface.
> 
> * DoIdentifyDeviceNVMeJMicron
> * DoIdentifyDeviceNVMeASMedia
> * DoIdentifyDeviceNVMeRealtek
> 
> are using vendors specific interface.

Sorry to bother you again, Hiyohiyo. I find the software of CrystalDiskInfo 8.8.7 does not read the hard disk of UASP (NVM Express) interface, but the software of CrystalDiskInfo 8.7.0 does read the hard disk of the two UASP (NVM Express). Where can the source code of CrystalDiskInfo 8.7.0 be downloaded? I want to debug through VS2017 and locate how the code gets its firmware information. Could you please tell me where CrystalDiskInfo 8.7.0 source code can be downloaded?

hiyohiyo, 2020-09-22:
You can download old version from 

- CrystalDiskInfo
 https://osdn.net/projects/crystaldiskinfo/releases/73507
- CrystalDiskInfo 8.7.0 Source Code 
 https://osdn.net/projects/crystaldiskinfo/downloads/73319/CrystalDiskInfo8_7_0Src.zip/

Change Log
https://github.com/hiyohiyo/CrystalDiskInfo/commits/master

CrystalDiskInfo 8.8.7 detects USB-NVMe drives on my environment.
If you notice the cause, please tell me it.

hiyohiyo, 2020-09-22:
I would like you to test several versions. CrystalDiskInfo detects your external NVMe SSD or not.

8.8.7  https://osdn.net/projects/crystaldiskinfo/downloads/73507/CrystalDiskInfo8_8_7.zip/
8.8.0  https://osdn.net/projects/crystaldiskinfo/downloads/73507/CrystalDiskInfo8_8_0.zip/
8.7.0  https://osdn.net/projects/crystaldiskinfo/downloads/73319/CrystalDiskInfo8_7_0.zip/

8.8.0 Alpha8  https://crystalmark.info/download/zz/CrystalDiskInfo8_8_0Alpha8.zip
8.8.0 Alpha9  https://crystalmark.info/download/zz/CrystalDiskInfo8_8_0Alpha9.zip

lwei2, 2020-09-23:
> You can download old version from
>
> * CrystalDiskInfo
>   https://osdn.net/projects/crystaldiskinfo/releases/73507
> * CrystalDiskInfo 8.7.0 Source Code [...]

Thank you very much.

lwei2, 2020-09-23:
> I would like you to test several versions. CrystalDiskInfo detects your external NVMe SSD or not.
>
> 8.8.7 https://osdn.net/projects/crystaldiskinfo/downloads/73507/CrystalDiskInfo8_8_7.zip/
> 8.8.0 [...]

Thank you very much.

lwei2, 2020-09-23:
> I would like you to test several versions. CrystalDiskInfo detects your external NVMe SSD or not.
>
> 8.8.7 https://osdn.net/projects/crystaldiskinfo/downloads/73507/CrystalDiskInfo8_8_7.zip/
> 8.8.0 [...]

I call DoIdentifyDeviceNVMeJMicron interface to get out of the hard drive firmware information NVMe interface. Hiyohiyo, thanks again for your help!

closed 2020-09-26 by hiyohiyo