hiyohiyo/CrystalDiskInfo #105: Understanding E-Mail Settings, Subject & Body structure

ertankucukoglu, 2020-12-06:
Hello,

I have a NVMe drive. It has "Critical Warning", "Available Spare" and "Percentage used" I want to have these information e-mailed to me when any of them changed.
[image]

I simply could not understand by reading examples provided on manual. 
https://crystalmark.info/en/software/crystaldiskinfo/crystaldiskinfo-alert-mail/

Do they already contain that data I want? If I have to add something in them and what it should be?

To be honest, I did not have a clear understanding of the structure for subject and body. I fail to find any other source.

I appreciate any help.

Thanks & Regards,
Ertan

hiyohiyo, 2020-12-08:
Critical Warning: No support
Available Spare: No support
Percentage used: No support

I will add Critical Warning support for CrystalDiskInfo 8.9.0.
Others will be supported for future release.

closed 2020-12-08 by hiyohiyo

ertankucukoglu, 2021-01-23:
Version 8.10.0 is released.

It is still not listed "Critical Warning" in below link. Is it because web site not updated, yet?
https://crystalmark.info/en/software/crystaldiskinfo/crystaldiskinfo-event-log/

Thanks for clarification.

Regards,
Ertan Küçükoğlu

hiyohiyo, 2021-02-07:
Sorry, I did not add individual support for Critical Warning.

CrystalDiskInfo reports "Bad" health status when Critical Warning sets a flag.