hiyohiyo/CrystalDiskInfo #111: SCSI/ATA Translation to SAT'Problem

lwei2, 2021-01-15:
Hi, hiyohiyo, In SCSI/ATA Translation to SAT,why comment out the sptwb.databuf [510]! = 0 xa5?

hiyohiyo, 2021-01-17:
A part of USB-SATA bridge chip returns "sptwb.databuf [510] != 0xA5".
If it checks strictly, it may not display anything...

closed 2021-01-17 by hiyohiyo

lwei2, 2021-01-18:
> A part of USB-SATA bridge chip returns "sptwb.databuf [510] != 0xA5".
> If it checks strictly, it may not display anything...
Hi,hiyohiyo,when  (I CrystalDiskInfo was used to read the hard disk of USBS-SATA bridge chip, the phenomenon of stalling appeared.Therefore, I would like to ask what might be the cause of this problem?

lwei2, 2021-01-18:
> A part of USB-SATA bridge chip returns "sptwb.databuf [510] != 0xA5".
> If it checks strictly, it may not display anything...
Hi,hiyohiyo,when I used CrystalDiskInfo to read the hard disk of USBS-SATA bridge chip, the phenomenon of stalling appeared.Therefore, I would like to ask what might be the cause of this problem?