hiyohiyo/CrystalDiskInfo #113: Impove Crucial BX500 (Micron) support

fuwa200fuwa, 2021-01-22:
```
F9 = 249 = Total Refresh ISP Count
FA = 250 = Total Do RefCal Count
FB = 251 = Total NAND Read Plane Count (Low 4Bytes)
FC = 252 = Total NAND Read Plane Count (High 4Bytes)
FD = 253 = Total Block ReMap Pass Count
FE = 254 = Total Background Scan Over Limit Count
DF = 223 = Total Background Scan
```

1\. CrystalDiskInfo 8.9.0a 2020/12/15

2\. Crucial Storage Executive 6.09.092020.06

<details><summary>3. msecli -S -V -n DriveN (v6.09.092020.06)</summary>

```
Device Name  : Drive4

 ID   DESCRIPTION                    FLAG  VALUE WORST RAW_DATA        Threshold

 1    Raw Read Error Rate            47    100   100   0               0  
 5    ReAllocated NAND Block Count   50    100   100   0               10 
 9    Power-On Hours Count           50    100   100   7               0  
 12   Power Cycle Count              50    100   100   2               0  
 171  Program Fail Count             50    100   100   0               0  
 172  Erase Fail Count               50    100   100   0               0  
 173  Average Block Erase Count      50    99    99    8               0  
 174  UnExpected Power-Loss Count    50    100   100   1               0  
 180  Remaining Reserved Block       51    100   100   39              0  
      Count                         
 183  Sata Interface Downshift       50    100   100   0               0  
 184  End-to-End Data Errors         50    100   100   0               0  
      corrected                     
 187  Reported UnCorrectable Errors  50    100   100   0               0  
 194  EnClosure Temperature          34    68    61    32 < 30 / 39  > 0  
							  <Min / Max > 
 196  ReAllocation Event Count       50    100   100   0               0  
 197  Current Pending ECC Count      50    100   100   0               0  
 198  SMART Off-line Scan            48    100   100   0               0  
      Uncorrectable Errors          
 199  Ultra-DMA CRC Error Count      50    100   100   0               0  
 202  Percent Lifetime Remaining     48    99    99    1               1  
 206  Write Error Rate               14    100   100   0               0  
 210  RAIN Successful Recovery       50    100   100   0               0  
       Page Count                   
 223  Total Background Scan          50    100   100   0               0  
      Total Background Scan         
 246  Cumulative Host Write          50    100   100   619598279       0  
      Sector Count                  
 247  Cumulative Host Write          50    100   100   19362446        0  
      Sector Count                  
 248  Total NAND Pages Written       50    100   100   10860512        0  
      Count                         
 249  Total Refresh ISP Count        50    100   100   0               0  
 250  Total Do RefCal Count          50    100   100   0               0  
 251  Total NAND Read Plane Count    50    100   100   257723463       0  
       (Low 4Bytes)                 
 252  Total NAND Read Plane Count    50    100   100   0               0  
       (High 4Bytes)                
 253  Total Block ReMap Pass Count   50    100   100   0               0  
 254  Total Background Scan          50    100   100   0               0  
       Over Limit Count             

Device Name  : Drive4
Verbose SMART data is retrived successfully
CMD_STATUS   : Success 
STATUS_CODE  : 0 

Copyright (C) 2020 Micron Technology, Inc.
```
</details>

[image]</details>

[image]

Only Crucial/Micron issue found: https://github.com/hiyohiyo/CrystalDiskInfo/issues/78

hiyohiyo, 2021-02-07:
Hi,

I added these S.M.A.R.T. attributes in the CrystalDiskInfo 8.11.0 Beta1.
https://crystalmark.info/download/zz/CrystalDiskInfo8_11_0Beta1.zip

Please test it and report the result.

Best regards,
hiyohiyo

F9 = 249 = Total Refresh ISP Count
FA = 250 = Total Do RefCal Count
FB = 251 = Total NAND Read Plane Count (Low 4Bytes)
FC = 252 = Total NAND Read Plane Count (High 4Bytes)
FD = 253 = Total Block ReMap Pass Count
FE = 254 = Total Background Scan Over Limit Count
DF = 223 = Total Background Scan

1. CrystalDiskInfo 8.9.0a 2020/12/15

2. Crucial Storage Executive 6.09.092020.06

3. msecli -S -V -n DriveN (v6.09.092020.06)
Device Name  : Drive4

 ID   DESCRIPTION                    FLAG  VALUE WORST RAW_DATA        Threshold

 1    Raw Read Error Rate            47    100   100   0               0  
 5    ReAllocated NAND Block Count   50    100   100   0               10 
 9    Power-On Hours Count           50    100   100   7               0  
 12   Power Cycle Count              50    100   100   2               0  
 171  Program Fail Count             50    100   100   0               0  
 172  Erase Fail Count               50    100   100   0               0  
 173  Average Block Erase Count      50    99    99    8               0  
 174  UnExpected Power-Loss Count    50    100   100   1               0  
 180  Remaining Reserved Block       51    100   100   39              0  
      Count                         
 183  Sata Interface Downshift       50    100   100   0               0  
 184  End-to-End Data Errors         50    100   100   0               0  
      corrected                     
 187  Reported UnCorrectable Errors  50    100   100   0               0  
 194  EnClosure Temperature          34    68    61    32 < 30 / 39  > 0  
							  <Min / Max > 
 196  ReAllocation Event Count       50    100   100   0               0  
 197  Current Pending ECC Count      50    100   100   0               0  
 198  SMART Off-line Scan            48    100   100   0               0  
      Uncorrectable Errors          
 199  Ultra-DMA CRC Error Count      50    100   100   0               0  
 202  Percent Lifetime Remaining     48    99    99    1               1  
 206  Write Error Rate               14    100   100   0               0  
 210  RAIN Successful Recovery       50    100   100   0               0  
       Page Count                   
 223  Total Background Scan          50    100   100   0               0  
      Total Background Scan         
 246  Cumulative Host Write          50    100   100   619598279       0  
      Sector Count                  
 247  Cumulative Host Write          50    100   100   19362446        0  
      Sector Count                  
 248  Total NAND Pages Written       50    100   100   10860512        0  
      Count                         
 249  Total Refresh ISP Count        50    100   100   0               0  
 250  Total Do RefCal Count          50    100   100   0               0  
 251  Total NAND Read Plane Count    50    100   100   257723463       0  
       (Low 4Bytes)                 
 252  Total NAND Read Plane Count    50    100   100   0               0  
       (High 4Bytes)                
 253  Total Block ReMap Pass Count   50    100   100   0               0  
 254  Total Background Scan          50    100   100   0               0  
       Over Limit Count             

Device Name  : Drive4
Verbose SMART data is retrived successfully
CMD_STATUS   : Success 
STATUS_CODE  : 0 

Copyright (C) 2020 Micron Technology, Inc.

Only Crucial/Micron issue found: #78

fuwa200fuwa, 2021-02-16:
Sorry, I didn't track this issue and v8.10.0 ("Improved Micron/Crucial SATA SSDs support").
Tried 8.10.0 and this 8.11.0 Beta1. These SMART parameters are recognized correctly in both versions.
[image]
Thanks.

closed 2021-02-20 by hiyohiyo