hiyohiyo/CrystalDiskInfo #125: CrystalDiskInfo 8.12.2 Shows no drives

JKZuchel, 2021-07-01:
Hello, I have installed crystal disk info 8.12.2, the issue is that it won’t show any internal disks, but it does recognize external drives.
All my drives are connected through SATA to my motherboard (Asus Prime X570-P)
Please tell me If there is any other info required.
System info:
Windows 10 Pro (21H1 - 19043.1081)
AMD R5 3600XT (Cpu)
AMD rx570 (Gpu)
*Picture 1 has no external drives connected - Picture 2 has 1 external drive connected

[image]
[image]

JKZuchel, 2021-07-02:
I like to comment that I have updated the bios for the motherboard, the version installed now is 4002 details. [here](https://www.asus.com/us/Motherboards-Components/Motherboards/All-series/PRIME-X570-P/HelpDesk_BIOS/)

beerisgood, 2021-07-02:
Did you use software like AV which block program features?
Windows Ransomware protection can also block stuff too, but you get a message about.

JKZuchel, 2021-07-07:
> 
> 
> Did you use software like AV which block program features?
> Windows Ransomware protection can also block stuff too, but you get a message about.

Could you explain more?

mikunya, 2021-08-14:
same problem with the x570 motherboard. 
X570 I AORUS PRO WIFI

hiyohiyo, 2021-08-15:
CrystalDiskInfo does not support AMD AHCI/NVMe/RAID driver.
Please use Microsoft AHCI/NVMe driver.

closed 2021-08-15 by hiyohiyo