hiyohiyo/CrystalDiskInfo #129: Micron 1100 SATA SSD: Total Host Reads/Writes converted wrong

iisti, 2021-08-11:
​I noticed that CrystalDiskInfo 8.12.5 is probably converting Micron 1100 SATA SSD disk's Total Host Reads/Writes wrong.

CrystalDiskInfo 8.12.5 shows these numbers for my disk.
~~~
Host Reads  : 1168593554 GB
Host Writes : 1785273526 GB
~~~

S.M.A.R.T info
~~~
F1 100 100 __0 000D4D2396CB Total LBA Write
F2 100 100 __0 0008B4EA5255 Total LBA Read
~~~

After converting HEX to DEC
~~~
000D4D2396CB = 57128752843
0008B4EA5255 = 37394993749
~~~

I also checked with `smartmontools` the Total LBAs.
~~~
241 Total_LBAs_Written      0x0033   100   100   000    Pre-fail  Always       -       57128815318
242 Total_LBAs_Read         0x0033   100   100   000    Pre-fail  Always       -       37395014575
~~~

I converted the LBAs with this calculator.
https://www.virten.net/2016/12/ssd-total-bytes-written-calculator/

These numbers are a bit more believable.
~~~
Total_LBAs_Written  57128815318 = 27241 GB
Total_LBAs_Read     37395014575 = 17831 GB
~~~

It seems that the CrystalDiskInfo has error in the conversion.

[micron1100_sata_ssd_DiskInfo64.log](https://github.com/hiyohiyo/CrystalDiskInfo/files/6969271/micron1100_sata_ssd_DiskInfo64.log)
[micron1100_sata_ssd_smartctl_a_output.txt](https://github.com/hiyohiyo/CrystalDiskInfo/files/6969272/micron1100_sata_ssd_smartctl_a_output.txt)
[micron1100_sata_ssd_text_copy.txt](https://github.com/hiyohiyo/CrystalDiskInfo/files/6969273/micron1100_sata_ssd_text_copy.txt)

hiyohiyo, 2021-08-16:
Maybe fixed by CrystalDiskInfo 8.12.6.

closed 2021-08-16 by hiyohiyo

iisti, 2021-08-16:
I tested with 8.12.6 and can confirm that issue has been fixed.
[micron1100_sata_ssd_text_copy_fixed.txt](https://github.com/hiyohiyo/CrystalDiskInfo/files/6992023/micron1100_sata_ssd_text_copy_fixed.txt)

commit d1dc0814 (2021-08-16): CrystalDiskInfo 8.12.6

- Improved NVMe SSD health status report
- Improved Micron SSD support

Maybe fixed #129