hiyohiyo/CrystalDiskInfo #154: About software starting speed

haobinnan, 2021-10-12:
I found that the new version starts very slowly (about a year). It is not as fast as the previous version.

hiyohiyo, 2021-12-06:
Sorry, too late reply.

I would like you to send DiskInfo.log of slow version and fast version.

How to create DiskInfo.log.
https://crystalmark.info/en/software/crystaldiskinfo/crystaldiskinfo-contact/

haobinnan, 2021-12-07:
00000000 8.13.0
00000000 2021/11/26
00000016 CreateMutex
00000032 AfxOleInit()
00000032 AfxOleGetMessageFilter()->SetMessagePendingDelay
00000047 AfxOleGetMessageFilter()->EnableNotRespondingDialog(FALSE)
00000047 AfxOleGetMessageFilter()->EnableBusyDialog(FALSE)
00000047 CDiskInfoDlg::CDiskInfoDlg - START
00000047 CDiskInfoDlg::CDiskInfoDlg - CENTER
00000047 CDiskInfoDlg::CDiskInfoDlg - END
00000047 dlg.DoModal()
00000422 InitAta
00000422 CAtaSmart::Init WMI on - Start
00000422 CoInitializeSecurity()
00000438 CoCreateInstance()
00000438 ConnectServer()
00000453 CoSetProxyBlanket()
00000453 OK:WMI Init
00002250 OK:Win32_IDEController
00002282 OK:Win32_SCSIController
00002282 --ControllerMap--
 + Intel(R) 300 Series Chipset Family SATA AHCI Controller [ATA]
   - PIONEER DVD-RW  DVR-219L
   - ST2000DM008-2FR102
   - ST4000NM000A-2HZ100
 - Microsoft 存储空间控制器 [SCSI]
 + Intel(R) SSD Pro 7600p/760p/E 6100p Series [SCSI]
   - INTEL SSDPEKKW256G8

00002297 OK:Win32_USBController
00002297 WakeUp(0)
00002313 WakeUp(1)
00002594 WakeUp(2)
00002594 DoIdentifyDeviceCsmi
00002610 asi.Model.IsEmpty() || asi.FirmwareRev.IsEmpty()
00002610 DoIdentifyDeviceCsmi
00002610 GetSmartAttributeCsmi
00002657 FillSmartData
00002657 GetSmartAttributeCsmi
00002688 FillSmartData
00002688 GetSmartThresholdCsmi
00002688 FillSmartThreshold
00002688 DoIdentifyDeviceCsmi
00002703 asi.Model.IsEmpty() || asi.FirmwareRev.IsEmpty()
00002703 DoIdentifyDeviceCsmi
00002703 asi.Model.IsEmpty() || asi.FirmwareRev.IsEmpty()
00002703 DoIdentifyDeviceCsmi
00002703 GetSmartAttributeCsmi
00002703 FillSmartData
00002703 GetSmartAttributeCsmi
00002703 FillSmartData
00002703 GetSmartThresholdCsmi
00002719 FillSmartThreshold
00002719 DoIdentifyDeviceCsmi
00002719 asi.Model.IsEmpty() || asi.FirmwareRev.IsEmpty()
00002719 DO:SELECT * FROM Win32_DiskDrive
00002719 OK1:SELECT * FROM Win32_DiskDrive
00002797 while(pEnumCOMDevs ...
00002797 diskSize:256052966400
00002797 deviceId:\\.\PHYSICALDRIVE0
00002797 model:INTEL SSDPEKKW256G8
00002797 mediaType:Fixed hard disk media
00002797 interfaceTypeWmi:SCSI
00002797 pnpDeviceId:SCSI\DISK&VEN_NVME&PROD_INTEL_SSDPEKKW25\5&34F18CBF&1&000000
00002797 DO:GetDiskInfo pd=0, sp=0, st=0, mt=fixed hard disk media
00002797 INTERFACE_TYPE_NVME
00002797 flagTarget && GetDiskInfo
00002813 GetDiskInfo
00002813 DoIdentifyDeviceNVMeStorageQuery
00002813 AddDiskNVMe - CMD_TYPE_NVME_STORAGE_QUERY
[... 104 lines cut ...]
00003047 Z:\
00003047 Drive Letter Mapping - != DRIVE_FIXED
00003047 Drive Letter Mapping - End
00003047 C
00003063 D
00003063 M
00003063 P
00003063 CAtaSmart::Init - Complete
00003063 SaveSmartInfo(i)
00003391 SaveSmartInfo(i)
00003922 SaveSmartInfo(i)
00004313 CheckResident()
00004313 SetTimer(TIMER_SET_POWER_ON_UNIT, 130000, 0)
00004313 AutoAamApmAdaption()
00004313 UpdateShareInfo()
00004313 RegisterDeviceNotification
00004313 InitDialogComplete
00004313 RestorePos()
00004328 CheckStartup
00004328 CheckHideSerialNumber
00004328 ChangeZoomType
00004328 UpdateDialogSize
00004578 ChangeLang
00004641 CheckPage()
00004641 ShowWindow(SW_SHOW)

hiyohiyo, 2021-12-07:
Thank you for log data.

Boot time is about 4.6 seconds. It may be normal.
I would like you to send the previous and fast boot version log.

haobinnan, 2021-12-08:
00000000 8.13.0
00000031 2021/11/26
00000062 CreateMutex
00000078 AfxOleInit()
00000093 AfxOleGetMessageFilter()->SetMessagePendingDelay
00000109 AfxOleGetMessageFilter()->EnableNotRespondingDialog(FALSE)
00000125 AfxOleGetMessageFilter()->EnableBusyDialog(FALSE)
00000140 CDiskInfoDlg::CDiskInfoDlg - START
00000172 CDiskInfoDlg::CDiskInfoDlg - CENTER
00000187 CDiskInfoDlg::CDiskInfoDlg - END
00000218 dlg.DoModal()
00000265 InitAta
00000297 CAtaSmart::Init WMI on - Start
00000297 CoInitializeSecurity()
00000328 CoCreateInstance()
00000343 ConnectServer()
00000359 CoSetProxyBlanket()
00000375 OK:WMI Init
00001500 OK:Win32_IDEController
00001578 OK:Win32_SCSIController
00001593 --ControllerMap--
 - Microsoft 存储空间控制器 [SCSI]
 + Intel(R) Chipset SATA/PCIe RST Premium Controller [SCSI]
   - Intel Raid 0 Volume
   - 通用软件组件
 + Intel(R) SSD Pro 7600p/760p/E 6100p Series [SCSI]
   - INTEL SSDPEKKW256G8
 + 标准 NVM Express 控制器 [SCSI]
   - WDS100T3X0C-00SJG0

00001672 OK:Win32_USBController
00001687 WakeUp(0)
00001703 WakeUp(1)
00001718 WakeUp(2)
00001734 DoIdentifyDeviceCsmi
00001750 asi.Model.IsEmpty() || asi.FirmwareRev.IsEmpty()
00001765 DoIdentifyDeviceCsmi
00001781 asi.Model.IsEmpty() || asi.FirmwareRev.IsEmpty()
00001797 DoIdentifyDeviceCsmi
00001812 GetSmartAttributeCsmi
00001906 FillSmartData
00001922 GetSmartAttributeCsmi
00002015 FillSmartData
00002031 GetSmartThresholdCsmi
00002062 FillSmartThreshold
00002093 DoIdentifyDeviceCsmi
00002250 GetSmartAttributeCsmi
00002343 FillSmartData
00002375 GetSmartAttributeCsmi
00002468 FillSmartData
00002500 GetSmartThresholdCsmi
00002531 FillSmartThreshold
00002562 DO:SELECT * FROM Win32_DiskDrive
00002578 OK1:SELECT * FROM Win32_DiskDrive
00002593 while(pEnumCOMDevs ...
00002609 diskSize:4000784417280
00002625 deviceId:\\.\PHYSICALDRIVE2
00002640 model:Intel Raid 0 Volume
00002672 mediaType:Fixed hard disk media
00002703 interfaceTypeWmi:SCSI
00002718 pnpDeviceId:SCSI\DISK&VEN_INTEL&PROD_RAID_0_VOLUME\4&2CB270B9&0&060200
00002734 DO:GetDiskInfo pd=2, sp=2, st=2, mt=fixed hard disk media
00002750 flagTarget && GetDiskInfo
00002765 GetDiskInfo
00002781 DoIdentifyDevicePd(2, 0xA0) - 1
00002797 SendAtaCommandPd - IDENTIFY_DEVICE (ATA_PASS_THROUGH)
00002812 SendAtaCommandPd - IDENTIFY_DEVICE
00002843 AddDisk(2, 2, 2) - 6
00002859 while(pEnumCOMDevs ...
00002875 diskSize:256052966400
[... 100 lines cut ...]
00005312 Drive Letter Mapping - != DRIVE_FIXED
00005343 Drive Letter Mapping - End
00005375 D
00005390 C
00005422 CAtaSmart::Init - Complete
00005437 SaveSmartInfo(i)
00005468 SaveSmartInfo(i)
00005500 SaveSmartInfo(i)
00005531 SaveSmartInfo(i)
00005562 CheckResident()
00005593 SetTimer(TIMER_SET_POWER_ON_UNIT, 130000, 0)
00005609 AutoAamApmAdaption()
00005640 UpdateShareInfo()
00005656 RegisterDeviceNotification
00005687 InitDialogComplete
00005718 RestorePos()
00005750 CheckStartup
00005781 CheckHideSerialNumber
00005812 ChangeZoomType
00005828 UpdateDialogSize
00006031 ChangeLang
00006062 GetSmartAttributeCsmi
00006187 FillSmartData
00006265 CheckPage()
00006312 ShowWindow(SW_SHOW)

haobinnan, 2021-12-08:
I'm sorry, I cannot find the fast booting version. It seems that it has something to do with the wd nvme disks. I can guarantee the wd nvme disk is normal. I hope the CAtaSmart class can be optimized so that the boot speed can be improved.

hiyohiyo, 2021-12-08:
I would like to optimize CAtaSmart in the future release.

closed 2021-12-08 by hiyohiyo