hiyohiyo/CrystalDiskInfo #157: Cell change watcher — watching of selected cells' value changes

AlttiRi, 2021-11-27:
When I open CrystalDiskInfo I usually go through all my hard drives to check only some selected values, for example: `CRC Error Count` and `Reallocated Sector Count`.

It takes time, I have to do monotonous clicking and visually find the necessary cells, also I should remember the old value to find out is the value has changed.

The decision of the problem — a **cell change watcher**.

How it should works?

Just click/double-click on a table cell and now it will be watched. You can mark it with a colored  underline (or with right/left colored borderline), or make the value italic.

When the value in the cell is changed the value should be the bold colored (for example) until you mouse move over it, or reactivate the disk tab by a click.

If a disk has one or more watched cells are changed it should be marked, for example, with the bold colored disk letter, until you mouse move over the all changed cells, or reactivate the disk tab by a click.

The quick made visual demo in Paint:

[image]

Here I have 4 watched cells (they have a colored right border) on the first disk `C:`. One cell was changed (`B1`).
The third disk has one (or more) changed watched cells so it's bold color marked as the first one.

AlttiRi, 2021-11-27:
With this enhancement I can just open the program and quickly find out were some important for me changes, or no. 
No need to spend time to manually recheck all disks.

yfdyh000, 2021-11-30:
I prefer to use tooltips to display increments since the last refresh and this session. Highlight cell changes can use italics, underlined, bold, and so on, colors is unfriendly for color disorders.

AlttiRi, 2021-12-01:
Also colored-background can use as an indicator. Or dots like it's in messengers when you have a new message.

I want to open the program and, with one glance at the tabs, determine whether there have been any important changes or not.
Without checking every disk tab.