hiyohiyo/CrystalDiskInfo #161: Kingston SSD can return Caution (Available Spare)

SkyEmie, 2021-12-11:
Hi,

I noticed that some kingston ssd can display "Caution", while on the older versions it was not the case.
Version 8.13.0 :
[image]

And this:
[output.log](https://github.com/hiyohiyo/CrystalDiskInfo/files/7697105/output.log)

It looks like a problem with the "Available Spare"?
Value and the threshold seem to be identical. In others softwares, the value is 100.

Hope this will help you!
Emie

hiyohiyo, 2021-12-12:
Thank you for your report.
I will fix this problem in the next release.

hiyohiyo, 2021-12-14:
I will fix it. Please test version.
https://crystalmark.info/download/zz/CrystalDiskInfo8_13_1.zip

SkyEmie, 2021-12-14:
That sounds good I guess!
What was the origin of the problem?

Emie

[logs.txt](https://github.com/hiyohiyo/CrystalDiskInfo/files/7714263/logs.txt)

[image]

hiyohiyo, 2021-12-14:
Thank you for testing. I will release CrystalDiskInfo 8.3.1 ASAP.

[image]

closed 2021-12-14 by hiyohiyo