hiyohiyo/CrystalDiskInfo #162: SSD Wear Indicator [E7] mark disk as critical when value = 100 on Kingston SA400S37120G

joaofvieira, 2022-01-20:
It seems there is an error on health status when the Kingston SA400S37120G has 100% of health. In this case, the health result is 0% (BAD) but the parameter E7 is listed as OK with a current value of 100 and the worst value is also 100. 
[image]
On another drive (same model)  a bit less healthier (98%) the  health status is UKNOWN with 98%. The E7 value on the list is marked "OK" (as well) but in this case the current value is 2 and the worst is also 2.
[image]

I also tried other software (including kingston's) and both drives are ok, one 100% healthy and the other with 98%.

I've a script that run on all machines boot that checks for drive health based on CrystalDiskInfo output. Because of this problem, I keep version 8.1.0 deployed on the domain.

hiyohiyo, 2022-01-22:
I bought KINGSTON SA400S37480G and tested it.
But, it will be correct.

I will add workaround in the next release.

[image]

----------------------------------------------------------------------------
 (03) KINGSTON SA400S37480G
----------------------------------------------------------------------------
           Model : KINGSTON SA400S37480G
        Firmware : SAH20105
   Serial Number : 50026B7282A949DE
       Disk Size : 480.1 GB (8.4/137.4/480.1/480.1)
     Buffer Size : Unknown
     Queue Depth : 32
    # of Sectors : 937703088
   Rotation Rate : ---- (SSD)
       Interface : Serial ATA
   Major Version : ACS-3
   Minor Version : ACS-3 Revision 4
   Transfer Mode : SATA/600 | SATA/600
  Power On Hours : 0 hours
  Power On Count : 6 count
      Host Reads : 0 GB
     Host Writes : 0 GB
     NAND Writes : 0 GB
     Temperature : 25 C (77 F)
   Health Status : Good (100 %)
        Features : S.M.A.R.T., APM, NCQ, TRIM
       APM Level : 0000h [OFF]
       AAM Level : ----
    Drive Letter : 

-- S.M.A.R.T. --------------------------------------------------------------
ID Cur Wor Thr RawValues(6) Attribute Name
01 100 100 __0 000000000064 Uncorrectable Errors
09 100 100 __0 000000000000 Power On Hours
0C 100 100 __0 000000000006 Power Cycles
94 100 100 __0 000000000000 Vendor Unique
95 100 100 __0 000000000000 Vendor Unique
A7 100 100 __0 000000000000 Vendor Unique
A8 100 100 __0 000000000000 SATA PHY Error Count
A9 100 100 __0 000000000000 Vendor Unique
AA 100 100 _10 000000000000 Bad Block Count
AC 100 100 __0 000000000000 Vendor Unique
AD 100 100 __0 000000000000 Erase Count
B5 100 100 __0 000000000000 Vendor Unique
B6 100 100 __0 000000000000 Vendor Unique
BB 100 100 __0 000000000000 Uncorrectable ECC Errors
C0 100 100 __0 000000000003 Vendor Specific
C2 _25 _25 __0 001600190019 Temperature
C4 100 100 __0 000000000000 Reallocation Events
C7 100 100 __0 000000000000 SATA CRC Error
DA 100 100 __0 000000000000 CRC Error Count
E7 100 100 __0 000000000064 SSD Wear Indicator
E9 100 100 __0 000000000000 Lifetime Nand Writes
F1 100 100 __0 000000000000 Lifetime Writes from Host
F2 100 100 __0 000000000000 Lifetime Read from Host
F4 100 100 __0 000000000000 Average Erase Count
F5 100 100 __0 000000000001 Max Erase Count
F6 100 100 __0 000000000002 Total Erase Count

-- IDENTIFY_DEVICE ---------------------------------------------------------
        0    1    2    3    4    5    6    7    8    9
000: 0040 3FFF C837 0010 0000 0000 003F 0000 0000 0000
010: 3530 3032 3642 3732 3832 4139 3439 4445 2020 2020
020: 0000 0000 0000 5341 4832 3031 3035 4B49 4E47 5354
030: 4F4E 2053 4134 3030 5333 3734 3830 4720 2020 2020
040: 2020 2020 2020 2020 2020 2020 2020 8001 4000 2F00
[... 66 lines cut ...]
070: 00 00 00 00 00 00 00 00 00 00 AD 00 00 00 00 00
080: 00 00 00 00 00 00 B5 00 00 00 00 00 00 00 00 00
090: 00 00 B6 00 00 00 00 00 00 00 00 00 00 00 BB 00
0A0: 00 00 00 00 00 00 00 00 00 00 C0 00 00 00 00 00
0B0: 00 00 00 00 00 00 C2 00 00 00 00 00 00 00 00 00
0C0: 00 00 C4 00 00 00 00 00 00 00 00 00 00 00 C7 00
0D0: 00 00 00 00 00 00 00 00 00 00 DA 00 00 00 00 00
0E0: 00 00 00 00 00 00 E7 00 00 00 00 00 00 00 00 00
0F0: 00 00 E9 00 00 00 00 00 00 00 00 00 00 00 F1 00
100: 00 00 00 00 00 00 00 00 00 00 F2 00 00 00 00 00
110: 00 00 00 00 00 00 F4 00 00 00 00 00 00 00 00 00
120: 00 00 F5 00 00 00 00 00 00 00 00 00 00 00 F6 00
130: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
140: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
150: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
160: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
170: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
180: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
190: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
1A0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
1B0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
1C0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
1D0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
1E0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
1F0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 1C

joaofvieira, 2022-01-22:
Maybe this is a firmware issue of this SSD Drives... Some give this error and others don't.
I've already run kingston SSD Manager and there is no firmware update for this drives!
Thank you so much for you help (and for the time you will spend solving this issue).

Best Regards,
Joao

hiyohiyo, 2022-01-23:
Hi,
I added workaround for Current Value == 100 and Raw Value == 0 model.
But, if Raw Value == 1, CrystalDiskInfo reports 99%...

I would like you to test it.
https://crystalmark.info/download/zz/CrystalDiskInfo8_15_0Dev2.zip

And I would like you to send the screen shot of Kingston SSD manager.
[image]

[image]

joaofvieira, 2022-01-23:
Hi,

I've tested on a disk 100% healthy (at this time I don't have access to the computer that has a drive with 99% healthy), and the result is this:
[image]

The disk is 100% Healthy but is marked as "BAD".

On Kingston SSD Manager, the result remains:
[image]

Once again, thanks for your patience and for your time.

Regards,

Joao

joaofvieira, 2022-01-24:
Hi again,

I've tested the 98% healthy drive and in this case the result on CrystalDiskInfo is:

[image]

On Kingston SSD Manager the result is:

[image]

The drives firmware is not the same, but kingston's software doesn't have any firmware update for both drives!

Best regards,

Joao Vieira

hiyohiyo, 2022-01-25:
Hi,
Thank you for test.

I forgot to update health status check function.
I fixed and updated.

I would like you to test 8.15.0 Dev3.
https://crystalmark.info/download/zz/CrystalDiskInfo8_15_0Dev3.zip

joaofvieira, 2022-01-25:
Hi,

I've tried on both disks with different firmware versions (03070009 and SBFKB1E1).

On firmware 03070009 the result is now OK:
[image]

On firmware SBFKB1E1 the result remains Unknown but the E7 attribute is marked as caution:
[image]

It seems there is a need of a firmware based workaround for these drives!

Once again, thanks for your patience and for your time.

Joao

hiyohiyo, 2022-01-29:
I added FirmwareRev base workaround. It may work fine....
https://crystalmark.info/download/zz/CrystalDiskInfo8_15_0Beta1.zip

joaofvieira, 2022-02-05:
HI!

Sorry for my late answer to your reply, but this week was not easy.

I've tested this version on both firmware versions of the SSD disk and the results are these:

Firmware 03070009 - The E7 value is at 100% but returns BAD (0%) as final result.
[image]

Firmware SBFKB1E1 - The E7 value is at 2% (yellow) and still returns UKNOWN (2%) as final result.
[image]

Thank you so much for your time and effort trying to solve this issue.

Best regards,

João

hiyohiyo, 2022-02-07:
Thank you for test.
I will fix ASAP.

joaofvieira, 2022-02-09:
Hi!

I've tested the latest version on both firmware versions and the results are:

On firmware 03070009 the result is now ok:
[image]

On firmware SBFKB1E1, despite E7 value is ok, the final result is "Unknown" with the correct percentage:
[image]

Once again, thank you for your effort trying to solve this issues.

Best regards,

João

hiyohiyo, 2022-02-09:
Thank you for testing.

I would like you to send the result of File > Save (text) to hiyohiyo@crystalmark.info.
I think that SBFKB1E1 drive returns incorrect S.M.A.R.T. information, so CrystalDiskInfo shows Unknown health status.

Test version.
https://crystalmark.info/download/zz/CrystalDiskInfo8_15_1Dev4.zip

hiyohiyo, 2022-02-11:
I released CrystalDiskInfo 8.15.1.
It almost same 8.15.1 Dev3.

I think  that firmware SBFKB1E1 reports uncommon data. So, CrystalDiskInfo reports Unknown as health status.

joaofvieira, 2022-02-11:
Ok.
Thank you so much for all your work.

Hackitfree, 2024-09-22:
I have one  KINGSTON SA400S37240G  firmware  SBFKB1E1. 
 I bought KINGSTON SA400S37240G firmware 03070009 .
Aside from firmware 03070009 giving 100% error rate -health is shown as it should - 97% in my case - bought used 
I needed another one identical because I wanted to put them in  RAID 0 
Problem is the old one with firmware  SBFKB1E1 is much faster - writing 40GB file 240MB/s is the minimum speed. 
The new one firmware 03070009 speed drops to 70MB/s
So when I put them in RAID 0 both write at 70MB/s because of the slower one. That one is at 100% usage while writing  -the other one chills at 25% not being fully utilized 
Any idea why are they so different ? At first I thought the new one is fake - but seeing your posts you have the same firmware's as me I thought I ask, 
Are your SSDs behave same way ? (one is slower in long write speeds? ) 

I appreciate your time and any insight will be helpful .
Thanks in advance 👍