hiyohiyo/CrystalDiskInfo #165: AMD_RC2t7 plugin not loading

Sylvanas-1977, 2022-02-06:
copy pasted straight into CrystalDiskInfo root directory.
file properties gotten rid of block.
Executing CrystalDiskInfo 8.15.0 function > advanced feature > AMD raid > plugin [web] clicking this just go's to the web page.
Are the instructions written wrong ? would imagine it loading plugins via a plugin folder instead.
raid xpert says vision 9.3.0-00296

Hope smart status works again eventually.

hiyohiyo, 2022-02-06:
Thank you for your bug report.
I will create test version and fix this problem ASAP.

Sylvanas-1977, 2022-02-07:
Thanks im looking forward to trying again to see if it works, i dont even know if it even loads anything normally have to unblock files, is there a way to check if it loads the plugin atm ?

hiyohiyo, 2022-02-07:
I may fix the bug.
https://crystalmark.info/download/zz/CrystalDiskInfo8_15_1Dev2.zip

Please test and send log files to hiyohiyo@crystalmark.info

-	Result of “File > Save (Text)”
-	Result of Debug.log
https://crystalmark.info/en/software/crystaldiskinfo/crystaldiskinfo-contact/
- Edit DiskInfo.ini from DebugMode=0 to DebugMode=1.
- Run DiskInfo64.exe

Sylvanas-1977, 2022-02-07:
Indeed works again
----------------------------------------------------------------------------
CrystalDiskInfo 8.15.1 Dev2 (C) 2008-2022 hiyohiyo
                                Crystal Dew World: https://crystalmark.info/
----------------------------------------------------------------------------

    OS : Windows 11 Professional [10.0 Build 22000] (x64)
  Date : 2022/02/07 15:55:03

-- Controller Map ----------------------------------------------------------
 + Standard NVM Express Controller [SCSI]
   - Samsung SSD 980 PRO 2TB
 + AMD-RAID Controller [storport] [SCSI]
   - AMD-RAID Config Device
   - AMD-RAID Array 1  SCSI Disk Device
   - AMD-RAID Array 3  SCSI Disk Device
 - Controlefunctie voor opslagruimten van Microsoft [SCSI]
 + AMD-RAID Bottom Device [SCSI]
   + AMD-RAID Controller [storport]
     - AMD-RAID Array 2  SCSI Disk Device
 - Xvdd SCSI Miniport [SCSI]
 - Microsoft VHD Loopback-controller [SCSI]
 + AMD-RAID Bottom Device [SCSI]
   - AMD-RAID Controller [storport]

-- Disk List ---------------------------------------------------------------
 (01) Samsung SSD 980 PRO 2TB : 2000,3 GB [0/0/0, sq] - nv
 (02) Samsung SSD 860 EVO 1TB : 1000,2 GB [1/0/0, rc] - sg
 (03) Samsung SSD 860 EVO 1TB : 1000,2 GB [1/0/0, rc] - sg
 (04) WDC WD40EZRZ-22GXCB0 : 4000,8 GB [2/0/0, rc]
 (05) CT2000MX500SSD1 : 2000,4 GB [3/0/0, rc] - m2
 (06) CT2000MX500SSD1 : 2000,4 GB [3/0/0, rc] - m2

----------------------------------------------------------------------------
 (01) Samsung SSD 980 PRO 2TB
----------------------------------------------------------------------------
           Model : Samsung SSD 980 PRO 2TB
        Firmware : 5B2QGXA7
   Serial Number : S69ENF0R849310V
       Disk Size : 2000,3 GB
       Interface : NVM Express
        Standard : NVM Express 1.3
   Transfer Mode : PCIe 4.0 x4 | PCIe 4.0 x4
  Power On Hours : 107 uren
  Power On Count : 26 keer
      Host Reads : 5044 GB
     Host Writes : 4340 GB
     Temperature : 33 C (91 F)
   Health Status : Goed (100 %)
        Features : S.M.A.R.T., TRIM, VolatileWriteCache
    Drive Letter : C:

-- S.M.A.R.T. --------------------------------------------------------------
ID RawValues(6) Attribute Name
01 000000000000 Kritieke Waarschuwing
02 000000000132 Composite Temperatuur
03 000000000064 Beschikbare Reserve's
04 00000000000A Beschikbare reserve drempel
05 000000000000 Percentage Gebruikt
06 000000A16F6C Data Units Lezen
07 0000008AE15D Data Units Schrijven
08 000003678212 Host Lees Opdrachten
09 00000305CAC6 Host Schrijf Opdrachten
0A 0000000000E1 Beheer Bezet Tijd
0B 00000000001A Power Cycles
0C 00000000006B Bedrijfs Uren
0D 000000000000 Onveilige Afsluiten
0E 000000000000 Media en Data Integriteit Fouten
0F 000000000000 Fout Informatie Logboeken

[... 765 lines cut ...]
070: 00 64 64 00 00 00 00 00 00 00 B8 32 00 64 64 00
080: 00 00 00 00 00 00 BB 32 00 64 64 00 00 00 00 00
090: 00 00 C2 22 00 4C 29 18 00 00 00 3B 00 00 C4 32
0A0: 00 64 64 00 00 00 00 00 00 00 C5 32 00 64 64 00
0B0: 00 00 00 00 00 00 C6 30 00 64 64 00 00 00 00 00
0C0: 00 00 C7 32 00 64 64 00 00 00 00 00 00 00 CA 30
0D0: 00 63 63 01 00 00 00 00 00 00 CE 0E 00 64 64 00
0E0: 00 00 00 00 00 00 D2 32 00 64 64 00 00 00 00 00
0F0: 00 00 F6 32 00 64 64 DB D4 29 FD 05 00 00 F7 32
100: 00 64 64 C8 A5 23 0C 00 00 00 F8 32 00 64 64 F8
110: 17 4E 0B 00 00 00 00 00 00 00 00 00 00 00 00 00
120: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
130: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
140: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
150: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
160: 00 00 00 00 00 00 00 00 00 00 80 00 00 00 00 7B
170: 03 00 01 00 02 1E 02 00 00 00 00 00 00 00 00 00
180: 00 00 4D 33 43 52 30 33 33 20 00 00 00 00 00 00
190: 53 4D 32 32 35 39 42 32 37 41 52 00 00 00 00 00
1A0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
1B0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
1C0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
1D0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
1E0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
1F0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 79

Sylvanas-1977, 2022-02-07:
cannot find the debug logs tho it just saves smartstatus like above, did set debugmode to 1

hiyohiyo, 2022-02-07:
Thank you!
It works fine. I will release 8.15.1 in the near future.

DiskInfo.ini 
From ----
[Setting]
DebugMode=0
To ----
[Setting]
DebugMode=1

closed 2022-02-08 by hiyohiyo