hiyohiyo/CrystalDiskInfo #167: Feature request: order list of disks by drive letter

Stan-C421, 2022-02-16:
As title, this is a feature request as I don't see anywhere to order the list of disks by drive letter.

Thank you.

[image]

hiyohiyo, 2023-02-19:
I apologize for making you wait a year.

I added sort by drive letter feature. Please test latest beta version.
https://crystalmark.info/download/zz/CrystalDiskInfo8_18_0Beta5.zip

Stan-C421, 2023-02-19:
I just tested the beta version and for me the disks are listed by drive letter. Thank you so much.
And no apologize required.

[image]