hiyohiyo/CrystalDiskInfo #177: Use 'AtaSmart' library on another function

camambeou, 2022-06-08:
I try to mod this project to split services and UI.
I write a .cpp to use 'AtaSmart' library like this:

CAtaSmart m_Ata{};
m_Ata.Init(1, 1, 0, 1, 1, 1);
ofstream myFile("C:\\ModelSerial.txt");
myFile << CT2A(m_Ata.vars[0].ModelSerial);
myFile.close();

ModelSerial of SATA disks is got.
NVMe disk is always null.
How can I fix that?

camambeou, 2022-06-08:
`CArray<ATA_SMART_INFO, ATA_SMART_INFO> vars` is always empty in this situation.

camambeou, 2022-06-10:
I guess thì code cause that:
`if (IsWindowsVersionOrGreaterFx(10, 0))
	{
		m_bAtaPassThrough = TRUE;
		m_bAtaPassThroughSmart = TRUE;
		m_bNVMeStorageQuery = TRUE;
	}
	else if (IsWindowsVersionOrGreaterFx(6, 0) || IsWindowsVersionOrGreaterFx(5, 2))
	{
		m_bAtaPassThrough = TRUE;
		m_bAtaPassThroughSmart = TRUE;
	}
	else if (IsWindowsVersionOrGreaterFx(5, 1))
	{
		if (IsWindowsVersionOrGreaterFx(5, 1, 2))
		{
			m_bAtaPassThrough = TRUE;
			m_bAtaPassThroughSmart = TRUE;
		}
	}`
Run on Windows 10, the above code leads to `else if (IsWindowsVersionOrGreaterFx(6, 0) || IsWindowsVersionOrGreaterFx(5, 2))`

camambeou, 2022-06-10:
I found my issue

closed 2022-06-10 by camambeou

zyyujq, 2023-02-12:
How to solve the problem? How to modify the code?

zyyujq, 2023-02-12:
It seems to be related to the operating system, WIN11 error, WIN7 can!

zyyujq, 2023-02-12:
#include<iostream> 
#include<fstream>
using namespace std;

CAtaSmart m_Ata{};
m_Ata.Init(1, 1, 0, 1, 1, 1);
ofstream myFile("C:\\ModelSerial.txt");
myFile << CT2A(m_Ata.vars[0].ModelSerial);
 myFile.close();

WIN7 passed!