hiyohiyo/CrystalDiskInfo #178: Kingston SSD health on certain Firmware.

wecome42, 2022-06-30:
Hello, I am using a Kingston SSD as my OS drive and I use CrystalDiskInfo all the time to check the health of the drives. However the drive shows Unknown 87%. According to the Kingston SSD Manager it shows 87% life so this is correct. I have updated the firmware and seems to be the same.

Firmware: SBFKB1H5
Model: KINGSTON SA400S37120G

[image]

hiyohiyo, 2022-06-30:
Please attach the text copy (Edit > Copy) and DiskInfo.log.
https://crystalmark.info/en/software/crystaldiskinfo/crystaldiskinfo-contact/

wecome42, 2022-07-01:
----------------------------------------------------------------------------
CrystalDiskInfo 8.17.3 (C) 2008-2022 hiyohiyo
                                Crystal Dew World: https://crystalmark.info/
----------------------------------------------------------------------------

    OS : Windows 10 Professional [10.0 Build 19043] (x64)
  Date : 2022/07/01 17:51:53

-- Controller Map ----------------------------------------------------------
 + Standard SATA AHCI Controller [ATA]
   - SAMSUNG HD321KJ
   - KINGSTON SA400S37120G
 - Microsoft Storage Spaces Controller [SCSI]

-- Disk List ---------------------------------------------------------------
 (01) SAMSUNG HD321KJ : 320.0 GB [0/0/0, pd1]
 (02) KINGSTON SA400S37120G : 120.0 GB [1/0/0, pd1] - ki

----------------------------------------------------------------------------
 (01) SAMSUNG HD321KJ
----------------------------------------------------------------------------
           Model : SAMSUNG HD321KJ
        Firmware : CP100-10
   Serial Number : *****************
       Disk Size : 320.0 GB (8.4/137.4/320.0/320.0)
     Buffer Size : 16384 KB
     Queue Depth : 32
    # of Sectors : 625142448
   Rotation Rate : Unknown
       Interface : Serial ATA
   Major Version : ATA8-ACS
   Minor Version : ATA8-ACS version 3b
   Transfer Mode : ---- | SATA/300
  Power On Hours : 7028 hours
  Power On Count : 592 count
     Temperature : 27 C (80 F)
   Health Status : Good
        Features : S.M.A.R.T., AAM, NCQ
       APM Level : ----
       AAM Level : FE00h [OFF]
    Drive Letter : E:

-- S.M.A.R.T. --------------------------------------------------------------
ID Cur Wor Thr RawValues(6) Attribute Name
01 100 100 _51 000000000000 Read Error Rate
03 100 100 _15 0000000015C0 Spin-Up Time
04 100 100 __0 000000000384 Start/Stop Count
05 253 253 _10 000000000000 Reallocated Sectors Count
07 253 253 _51 000000000000 Seek Error Rate
08 253 253 _15 000000000000 Seek Time Performance
09 100 100 __0 000000001B74 Power-On Hours
0A 253 253 _51 000000000000 Spin Retry Count
0B 253 100 __0 000000000000 Recalibration Retries
0C 100 100 __0 000000000250 Power Cycle Count
BB 253 253 __0 000000000000 Reported Uncorrectable Errors
BC 253 253 __0 000000000000 Command Timeout
BE _73 _53 __0 00000000001B Airflow Temperature
C2 157 _97 __0 00000000001B Temperature
C3 100 100 __0 0000001150C9 Hardware ECC recovered
C4 253 253 __0 000000000000 Reallocation Event Count
C5 253 253 __0 000000000000 Current Pending Sector Count
C6 253 253 __0 000000000000 Uncorrectable Sector Count
C7 200 200 __0 000000000000 UltraDMA CRC Error Count
C8 100 100 __0 000000000000 Write Error Rate
C9 253 100 __0 000000000000 Soft Read Error Rate
CA 253 253 __0 000000000000 Data Address Mark Error

-- IDENTIFY_DEVICE ---------------------------------------------------------
        0    1    2    3    4    5    6    7    8    9
000: 0040 3FFF C837 0010 8856 022A 003F 0000 0000 0000
[... 226 lines cut ...]
070: 00 00 00 00 00 00 00 00 00 00 AD 00 00 00 00 00
080: 00 00 00 00 00 00 B5 00 00 00 00 00 00 00 00 00
090: 00 00 B6 00 00 00 00 00 00 00 00 00 00 00 BB 00
0A0: 00 00 00 00 00 00 00 00 00 00 C0 00 00 00 00 00
0B0: 00 00 00 00 00 00 C2 00 00 00 00 00 00 00 00 00
0C0: 00 00 C4 00 00 00 00 00 00 00 00 00 00 00 C7 00
0D0: 00 00 00 00 00 00 00 00 00 00 DA 00 00 00 00 00
0E0: 00 00 00 00 00 00 E7 00 00 00 00 00 00 00 00 00
0F0: 00 00 E9 00 00 00 00 00 00 00 00 00 00 00 F1 00
100: 00 00 00 00 00 00 00 00 00 00 F2 00 00 00 00 00
110: 00 00 00 00 00 00 F4 00 00 00 00 00 00 00 00 00
120: 00 00 F5 00 00 00 00 00 00 00 00 00 00 00 F6 00
130: 00 00 00 00 00 00 00 00 00 00 F6 00 00 00 00 00
140: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
150: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
160: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
170: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
180: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
190: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
1A0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
1B0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
1C0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
1D0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
1E0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
1F0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 21

hiyohiyo, 2022-07-16:
Sorry, too late reply.
Thank you for log data.

All threshold data is 0. So, CrystalDiskInfo reports Unknown as health status.

F6 attribute is duplicated. It may buggy firmware...

F6 100 100 __0 00000007C1C8 Total Erase Count
F6 100 100 __0 00000007C1C8 Total Erase Count

closed 2022-07-16 by hiyohiyo