hiyohiyo/CrystalDiskInfo #180: Program steals focus while in the taskbar when you change power source

ponjadito23, 2022-07-15:
If you connect or disconnect your laptop while CrystalDiskInfo is in the taskbar and you have opened the GUI at least once, whatever program you're using will lose focus.

I discovered it was CrystalDiskInfo because when you press Alt+F4, CrystalDiskInfo would close and the problem no longer happened.

I'm using the latest version (8.17.3).

hiyohiyo, 2022-07-16:
Sorry, I did not understand the procedure to reproduce it.

Could you please give me a step-by-step procedure? Images or videos would make it easier to understand.

And please attach the text copy (Edit > Copy) and DiskInfo.log.
https://crystalmark.info/en/software/crystaldiskinfo/crystaldiskinfo-contact/

ponjadito23, 2022-07-19:
Sure, I'll attach the logs and a video showing the problem.

[DiskInfo64S.log](https://github.com/hiyohiyo/CrystalDiskInfo/files/9143804/DiskInfo64S.log)

[CrystalDiskInfo Edit Copy.txt](https://github.com/hiyohiyo/CrystalDiskInfo/files/9143854/CrystalDiskInfo.Edit.Copy.txt)

https://user-images.githubusercontent.com/17890647/179829291-63fadbe1-76ee-4913-b339-253a3fd7d582.mp4

hiyohiyo, 2022-07-22:
Thank you for your detailed explanation.
I will investigate this weekend to determine the cause.

hiyohiyo, 2022-07-23:
I could not reproduce the problem in my environment.
I have fixed the suspicious part and would appreciate it if you could test it.

https://crystalmark.info/download/zz/CrystalDiskInfo8_17_5Beta2.zip

-----
LRESULT CDiskInfoDlg::OnPowerBroadcast(WPARAM wParam, LPARAM lParam)
{
	switch(wParam)
	{
	case PBT_APMRESUMESUSPEND:
		SetTimer(TIMER_FORCE_REFRESH, 5 * 1000, 0);
		return TRUE;
		break;
	default:
		break;
	}

        // from TRUE to FALSE
	return FALSE;
}

ponjadito23, 2022-07-23:
Thanks. I've tested it and it keeps happening.
But now I'm curious if it only happens on this machine, I'll try on another.

ponjadito23, 2022-07-23:
Ok, I've tested the stable version on another computer and it doesn't happen there.

I've done extra tests on my main machine and I've noticed that the GUI window reloads when switching power.
I also checked the resource usage on Process Explorer and the memory seems to spike briefly even when I've not opened the GUI at least once, so no focus steal yet.
And all of this only seems to happen if explorer.exe is running.

I'll attach some screenshots of the mem spikes on both the Shizuku version I'm using and the beta you sent me.

[image]
[image]

hiyohiyo, 2022-07-24:
Thank you for your testing...
I am investigating but have not yet identified the cause.

Does this problem also occur when not resident?

hiyohiyo, 2022-07-25:
Thank you.
I will investigate resident-related processes.

hiyohiyo, 2023-02-19:
It may fix stealing focus problem.
https://crystalmark.info/download/zz/CrystalDiskInfo8_18_0Beta5.zip

I'd be happy to test it for you.

Reference: How to show a MFC dialog without stealing focus on the other window
https://stackoverflow.com/questions/2122278/how-to-show-a-mfc-dialog-without-stealing-focus-on-the-other-window

ponjadito23, 2023-02-22:
It still happens :(

I've previously tried debugging the program and was able to find the line of code that causes the focus steal. It seems that it executes every time I plug or unplug the computer, even if I don't open the UI even once so that it doesn't steal focus. 
I created two dumps before the program executes it when plugging/unplugging, the first one without opening the UI and the second one after opening and closing it once. 

[DiskInfo dumps.zip](https://github.com/hiyohiyo/CrystalDiskInfo/files/10804659/DiskInfo.dumps.zip)