hiyohiyo/CrystalDiskInfo #191: 0% Health on a brand new SSD

NekomataGami, 2022-11-21:
I recently bought this new SSD and CrystalDiskInfo is reading it as 0% health, I tried other disk monitoring software and its actually on 100% health. 
[image]

hiyohiyo, 2022-11-21:
Thank you for bug report.
I would like you to send the result of File > Save (text).

NekomataGami, 2022-11-21:
> Thank you for bug report. I would like you to send the result of File > Save (text).

Here's the text file.

[CrystalDiskInfo_20221122005204.txt](https://github.com/hiyohiyo/CrystalDiskInfo/files/10058475/CrystalDiskInfo_20221122005204.txt)

hiyohiyo, 2022-12-03:
I fixed this problem in 8.17.13.

closed 2022-12-03 by hiyohiyo