hiyohiyo/CrystalDiskInfo #199: Error disk unknown

luimen6, 2023-01-18:
Could you check this disk, please? LITEON L8H-256V2G-11 M.2 2280 256GB 
SMART Unknown.

luimen6, 2023-01-31:
@hiyohiyo Do you have any updates for me please?

luimen6, 2023-02-03:
[DiskInfo64.log](https://github.com/hiyohiyo/CrystalDiskInfo/files/10578007/DiskInfo64.log)

hiyohiyo, 2023-02-12:
If you send a bug report to me, please attach the text copy (Edit > Copy).

https://crystalmark.info/en/software/crystaldiskinfo/crystaldiskinfo-contact/

luimen6, 2023-02-12:
You mean that log file?
[DiskInfo64.log](https://github.com/hiyohiyo/CrystalDiskInfo/files/10716100/DiskInfo64.log)

luimen6, 2023-02-12:
@hiyohiyo 

And I sent you the text: https://github.com/hiyohiyo/CrystalDiskInfo/issues/199#issuecomment-1415639415

hiyohiyo, 2023-02-13:
@luimen6 

No, I would like you to send the result of "Menu > Edit > Copy" or "Menu > File > Save (Text)".

luimen6, 2023-02-17:
@hiyohiyo 

> ----------------------------------------------------------------------------
> CrystalDiskInfo 8.17.14 (C) 2008-2022 hiyohiyo
>                                 Crystal Dew World: https://crystalmark.info/
> ----------------------------------------------------------------------------
> 
>     OS : Windows 10 Professional [10.0 Build 19045] (x64)
>   Date : 2023/02/17 9:56:31
> 
> -- Controller Map ----------------------------------------------------------
>  + Intel(R) Chipset SATA/PCIe RST Premium Controller [SCSI]
>    - LITEON L8H-256V2G-11 M.2 2280 256GB
>    - Intel(R) Pinning Shell Extensions
>  - Microsoft-Controller für Speicherplätze [SCSI]
> 
> -- Disk List ---------------------------------------------------------------
>  (01) LITEON L8H-256V2G-11 M.2 2280 256GB : 256,0 GB [0/0/2, pd1] - px
> 
> ----------------------------------------------------------------------------
>  (01) LITEON L8H-256V2G-11 M.2 2280 256GB
> ----------------------------------------------------------------------------
>            Model : LITEON L8H-256V2G-11 M.2 2280 256GB
>         Firmware : F88110C
>    Serial Number : TW0MGNHV5508562NH5T1
>        Disk Size : 256,0 GB (8,4/137,4/256,0/----)
>      Buffer Size : Unbekannt
>      Queue Depth : 32
>     # of Sectors : 500118192
>    Rotation Rate : ---- (SSD)
>        Interface : Serial ATA
>    Major Version : ACS-2
>    Minor Version : ----
>    Transfer Mode : SATA/600 | SATA/600
>   Power On Hours : 14 Std.
>   Power On Count : 719 mal
>       Host Reads : 4049 GB
>      Host Writes : 2844 GB
>      Temperature : Unbekannt
>    Health Status : Unbekannt
>         Features : S.M.A.R.T., AAM, NCQ, TRIM, DevSleep, GPL
>        APM Level : ----
>        AAM Level : FE80h [ON]
>     Drive Letter : C:
> 
> -- S.M.A.R.T. --------------------------------------------------------------
> ID Cur Wor Thr RawValues(6) Attribute Name
> 05 100 100 __0 000000000000 Anz. wiederzugewiesener Sektoren
> 09 100 100 __0 00000000000E Betriebsstunden
> 0C 100 100 __0 0000000002CF Anz. Geräte-Einschaltvorgänge
> AF 100 100 __0 000000000000 Programmfehler (schlechtester Wert)
> B0 100 100 __0 000000000000 Löschfehler (schlechtester Wert)
> B1 100 100 __0 000000000007 Verschleißausgleich-Ereignisse
> B2 100 100 __0 000000000000 Anz. verwendeter reservierter Blöcke (schlechtester Wert)
> B3 100 100 __0 000000000000 Benutzte reservierte Blöcke
> B4 100 100 __0 000000000092 Unbenutzte reservierte Blöcke (gesamt)
> B5 100 100 __0 000000000000 Programmfehler
> B6 100 100 __0 000000000000 Löschfehler
> BB 100 100 __0 000000000000 Nicht korrigierbare Fehler
> C3 100 100 __0 000000000000 ECC-Rate
> F1 100 100 __0 00000001639C Host-Schreibvorgänge (gesamt)
> F2 100 100 __0 00000001FA3F Host-Lesevorgänge (gesamt)
> 
> -- IDENTIFY_DEVICE ---------------------------------------------------------
>         0    1    2    3    4    5    6    7    8    9
> 000: 0040 3FFF C837 0010 0000 0000 003F 0000 0000 0000
> 010: 5457 304D 474E 4856 3535 3038 3536 324E 4835 5431
> 020: 0000 0000 0000 4638 3831 3130 4320 4C49 5445 4F4E
> 030: 204C 3848 2D32 3536 5632 472D 3131 204D 2E32 2032
> 040: 3238 3020 3235 3647 4220 2020 2020 8002 4000 2F00
[... 66 lines cut ...]
> 070: 00 00 00 00 00 00 00 00 00 00 B6 00 00 00 00 00
> 080: 00 00 00 00 00 00 BB 00 00 00 00 00 00 00 00 00
> 090: 00 00 C3 00 00 00 00 00 00 00 00 00 00 00 F1 00
> 0A0: 00 00 00 00 00 00 00 00 00 00 F2 00 00 00 00 00
> 0B0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
> 0C0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
> 0D0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
> 0E0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
> 0F0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
> 100: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
> 110: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
> 120: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
> 130: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
> 140: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
> 150: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
> 160: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
> 170: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
> 180: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
> 190: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
> 1A0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
> 1B0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
> 1C0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
> 1D0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
> 1E0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
> 1F0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 F0

hiyohiyo, 2023-02-19:
I may fix it. Please test latest beta version.
https://crystalmark.info/download/zz/CrystalDiskInfo8_18_0Beta5.zip

luimen6, 2023-02-24:
@hiyohiyo 
I get health status: Unknown
[DiskInfo64.log](https://github.com/hiyohiyo/CrystalDiskInfo/files/10822319/DiskInfo64.log)

>  ----------------------------------------------------------------------------
> CrystalDiskInfo 8.18.0 Beta5 (C) 2008-2023 hiyohiyo
>                                 Crystal Dew World: https://crystalmark.info/
> ----------------------------------------------------------------------------
> 
>     OS : Windows 10 Professional [10.0 Build 19045] (x64)
>   Date : 2023/02/24 9:49:05
> 
> -- Controller Map ----------------------------------------------------------
>  + Intel(R) Chipset SATA/PCIe RST Premium Controller [SCSI]
>    - LITEON L8H-256V2G-11 M.2 2280 256GB
>    - Intel(R) Pinning Shell Extensions
>  - Microsoft-Controller für Speicherplätze [SCSI]
> 
> -- Disk List ---------------------------------------------------------------
>  (01) LITEON L8H-256V2G-11 M.2 2280 256GB : 256,0 GB [0/0/2, pd1] - px
> 
> ----------------------------------------------------------------------------
>  (01) LITEON L8H-256V2G-11 M.2 2280 256GB
> ----------------------------------------------------------------------------
>            Model : LITEON L8H-256V2G-11 M.2 2280 256GB
>         Firmware : F88110C
>    Serial Number : TW0MGNHV5508562NH5T1
>        Disk Size : 256,0 GB (8,4/137,4/256,0/----)
>      Buffer Size : Unknown
>      Queue Depth : 32
>     # of Sectors : 500118192
>    Rotation Rate : ---- (SSD)
>        Interface : Serial ATA
>    Major Version : ACS-2
>    Minor Version : ----
>    Transfer Mode : SATA/600 | SATA/600
>   Power On Hours : 14 hours
>   Power On Count : 732 count
>       Host Reads : 4135 GB
>      Host Writes : 2916 GB
>      Temperature : Unknown
>    Health Status : Unknown
>         Features : S.M.A.R.T., AAM, NCQ, TRIM, DevSleep, GPL
>        APM Level : ----
>        AAM Level : FE80h [ON]
>     Drive Letter : C:
> 
> -- S.M.A.R.T. --------------------------------------------------------------
> ID Cur Wor Thr RawValues(6) Attribute Name
> 05 100 100 __0 000000000000 Reallocated Sectors Count
> 09 100 100 __0 00000000000E Power-On Hours
> 0C 100 100 __0 0000000002DC Power Cycle Count
> AF 100 100 __0 000000000000 Program Fail Count (Worst Case)
> B0 100 100 __0 000000000000 Erase Fail Count (Worst Case)
> B1 100 100 __0 000000000008 Wear Leveling Count
> B2 100 100 __0 000000000000 Used Reserved Block Count (Worst Case)
> B3 100 100 __0 000000000000 Used Reserved Block Count
> B4 100 100 __0 000000000092 UnUsed Reserved Block Count (Total)
> B5 100 100 __0 000000000000 Program Fail Count
> B6 100 100 __0 000000000000 Erase Fail Count
> BB 100 100 __0 000000000000 Uncorrectable Error Count
> C3 100 100 __0 000000000000 ECC rate
> F1 100 100 __0 000000016C99 Total Host Writes
> F2 100 100 __0 0000000204ED Total Host Reads
> 
> -- IDENTIFY_DEVICE ---------------------------------------------------------
>         0    1    2    3    4    5    6    7    8    9
> 000: 0040 3FFF C837 0010 0000 0000 003F 0000 0000 0000
> 010: 5457 304D 474E 4856 3535 3038 3536 324E 4835 5431
> 020: 0000 0000 0000 4638 3831 3130 4320 4C49 5445 4F4E
[... 68 lines cut ...]
> 070: 00 00 00 00 00 00 00 00 00 00 B6 00 00 00 00 00
> 080: 00 00 00 00 00 00 BB 00 00 00 00 00 00 00 00 00
> 090: 00 00 C3 00 00 00 00 00 00 00 00 00 00 00 F1 00
> 0A0: 00 00 00 00 00 00 00 00 00 00 F2 00 00 00 00 00
> 0B0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
> 0C0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
> 0D0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
> 0E0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
> 0F0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
> 100: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
> 110: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
> 120: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
> 130: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
> 140: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
> 150: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
> 160: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
> 170: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
> 180: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
> 190: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
> 1A0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
> 1B0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
> 1C0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
> 1D0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
> 1E0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
> 1F0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 F0