hiyohiyo/CrystalDiskInfo #200: Stealing focus when starting with Windows

KarellenX, 2023-01-28:
Hello,

I have been a user of your software for many, many years. Thank you.

The issue I have is that when CrystalDiskInfo starts up with Windows, it steals focus.
I have Kodi running on my HTPC to autostart with Windows. Kodi starts, then a few minutes later CrystalDiskInfo starts and then sits on top of my running media centre. This is not recent, and has been this way for many years. But because it can't be controlled by remote control, I need to pull out the keyboard and close CrystalDiskInfo.

Is there anyway to prevent stealing focus? Maybe a setting to prevent this?

Thankyou

hiyohiyo, 2023-01-29:
Thank you for your report.

CrystalDiskInfo uses SetForegroundWindow() when it shows windows. So, I make test version which does not use  SetForegroundWindow().
https://github.com/hiyohiyo/CrystalDiskInfo/commit/84856b09056ddd8c0d6d2feb16b92b6290dec5b9

Please download Alpha version from below link. and test it.
https://crystalmark.info/download/zz/CrystalDiskInfo8_18_0Alpha.zip

KarellenX, 2023-01-31:
Hello @hiyohiyo 

My apologies for the delay in replying as I was a bit caught up yesterday.

I have tried installing the alpha version by running the **DiskInfo64.exe** in the zip file from your link but I receive an error during installation...

[image]

KarellenX, 2023-01-31:
Here is a short 48 second video that captures the startup of my HTPC (slow startup due to NextPVR server), Kodi starts (25sec) then CrystalDisk takes focus (44sec). 

From the time I select Restart to the end of the video there are no keyboard, mouse or other input made.

https://streamable.com/mzi2kb

hiyohiyo, 2023-01-31:
Please unzip all files.
DiskInfo64.exe required "CdiResource" files.

KarellenX, 2023-02-01:
Ok, thanks.

I believe I am misunderstanding what to do with this zip file. 

I have extracted it, double clicked the DiskInfo64.exe file and allowed Windows to install. I did this with the assumption that it was the installer file. But I don't think it is, because when I ran it, then restarted my pc, there was no change.

Am I instead meant to copy these files to the existing install?

hiyohiyo, 2023-02-01:
I prepare installer version. Please use it.
https://crystalmark.info/download/zz/CrystalDiskInfo8_18_0Alpha.exe

KarellenX, 2023-02-02:
Thank you for the installer.

I have uninstalled the previous version and installed the above Alpha version.

Unfortunately, there is no change and it still pops up over previously running window (Kodi) as shown in the video above.

hiyohiyo, 2023-02-02:
Do you enable resident and startup option?

[image]

I will investigate this issue this weekend.

KarellenX, 2023-02-02:
No, I did not enable Resident as I did not know what it was for.

I will do that when I return home later.
Thankyou.

hiyohiyo, 2023-02-12:
It may fix stealing focus problem.
https://crystalmark.info/download/zz/CrystalDiskInfo8_18_0Beta4.zip

I'd be happy to test it for you.

Reference: How to show a MFC dialog without stealing focus on the other window
https://stackoverflow.com/questions/2122278/how-to-show-a-mfc-dialog-without-stealing-focus-on-the-other-window

hiyohiyo, 2023-02-19:
It may fix stealing focus problem.
https://crystalmark.info/download/zz/CrystalDiskInfo8_18_0Beta5.zip

I'd be happy to test it for you.

Reference: How to show a MFC dialog without stealing focus on the other window
https://stackoverflow.com/questions/2122278/how-to-show-a-mfc-dialog-without-stealing-focus-on-the-other-window