hiyohiyo/CrystalDiskInfo #201: SSD Wear Indicator [E7] mark disk as critical on Kingston SA400S37120G

lehuymanhtan, 2023-02-09:
I just upgraded CrystalDiskInfo to the latest version and there seems to be a bug with the health status of Kingston SA400S37120G. In this case, the health result is bad but the parameter E7 is listed as OK with a current value of 99 and the worst value is also 99.

[image]

I also tried Kingston SSD Manager and the drive is ok.

[image]

hiyohiyo, 2023-02-12:
Thanks for your bug report.
I'll fix it in the next version!

hiyohiyo, 2023-02-19:
I fixed support for Kingston SA400S37120G. Please test latest beta version.
https://crystalmark.info/download/zz/CrystalDiskInfo8_18_0Beta5.zip

lehuymanhtan, 2023-02-20:
I've tested it and it works as expected. 
Here is the result:
[image]