hiyohiyo/CrystalDiskInfo #202: How to change AtaSmart in CrystalDiskInfo to DLL and call it to VB or C #

zyyujq, 2023-02-12:
Simply obtain the disk model and serial number, plan it as a simple DLL library, call it to VB or C #, and how to simplify AtaSmart in CrystalDiskInfo.

hiyohiyo, 2023-02-12:
There are plans to release a DLL version.
However, we have not determined when we will be able to release it.

closed 2023-02-12 by hiyohiyo

ftyszyx, 2024-09-19:
I wrapped CrystalDiskInfo into a DLL. Now, other languages or Windows applications can also accurately distinguish the disk serial number with the help of CrystalDiskInfo. I have open-sourced this project on GitHub. If you are interested in it, you can check it out and use it.
https://github.com/ftyszyx/CrystalDiskInfo_dll_lib

chmichael, 2024-09-19:
You should add the version of the CrystalDiskInfo included in the dll and release page.

ftyszyx, 2024-09-19:
@chmichael ok,I will add it