hiyohiyo/CrystalDiskInfo #203: Where is the error when referencing the CAtaSmart class?

zyyujq, 2023-02-13:
VS2019 creates a dialog box MFC, references CAtaSmart related classes, and compiles the program without errors.

In Windows 7, the administrator runs the program without errors.
In Windows 11, the administrator ran a program error.

The error report is as follows:
2019\....... \MSVC\14.29.30133\atlmfc\include\afxtempl.h Line:320

---
VC++ code:
Void CHDDSNDlg:: OnOK()//Calculation button
{
CAtaSmart m_ ATA;
m_ ATA.Init(TRUE, FALSE, NULL, TRUE, TRUE, FALSE);
CString strHDDSN = _ T("WS"); // m_ ATA.vars[0].SerialNumber;
CString strHDDName = m_ ATA.vars[0].Model;
SetDlgItemText(IDC_EDIT1, strHDDName);
SetDlgItemText(IDC_EDIT2,strHDDSN);
}

Code errors should be handled by CAtaSmart related classes. How to troubleshoot errors?

zyyujq, 2023-02-13:
**Save the following content as APP.exe.manifest.
In the input and output of the list tool on the VS2019 project property page, add the additional list file setting to APP.exe.manifest.**

---

APP.exe.manifest:

< ?xml version="1.0" encoding="UTF-8" standalone="yes"?>
< assembly xmlns="urn:schemas-microsoft-com:asm.v1" manifestVersion="1.0">
    < compatibility xmlns="urn:schemas-microsoft-com:compatibility.v1">
        < application>
            < !-- Windows 10 and Windows 11 -->
            < supportedOS Id="{8e0f7a12-bfb3-4fe8-b9a5-48fd50a15a9a}"/>
            < !-- Windows 8.1 -->
            < supportedOS Id="{1f676c76-80e1-4239-95bb-83d0f6d0da78}"/>
            < !-- Windows 8 -->
            < supportedOS Id="{4a2f28e3-53b9-4441-ba9c-d69d4a4a6e38}"/>
            < !-- Windows 7 -->
            < supportedOS Id="{35138b9a-5d96-4fbd-8e2d-a2440225f93a}"/>
            < !-- Windows Vista -->
            < supportedOS Id="{e2011457-1546-43c5-a5fe-008deee3d3f0}"/> 
        < /application>
    < /compatibility>
< /assembly>

hiyohiyo, 2023-02-14:
Please rebuild original version before customizing...

closed 2023-04-03 by hiyohiyo