hiyohiyo/CrystalDiskInfo #215: Not working on hp proliant ml350 g6

Felzow47, 2023-03-30:
[image]

TP555, 2023-03-31:
Hi

And with the 9.0.0 Beta 4  ?

https://crystalmark.info/en/

Cheers  !

ShigeakiAsai, 2023-04-01:
Hello, Felzow47

I think developper needs more infomation about this issue.
Please attach the text copy (Edit > Copy) and DiskInfo.log.

BTW
According to the specification of hp proliant ml350 g6,
(https://support.hpe.com/hpesc/public/docDisplay?docLocale=en_US&docId=c01713519)
default storage controller is "Smart Array P410i controller".

Do you know which storage controller is used in your PC?

Thank you!

Felzow47, 2023-04-01:
> Hello, Felzow47
>
> I think developper needs more infomation about this issue. Please attach the text copy (Edit > Copy) and DiskInfo.log.
>
> BTW According to the specification of hp proliant ml350 g6, [...]

----------------------------------------------------------------------------
CrystalDiskInfo 8.17.14 (C) 2008-2022 hiyohiyo
                                Crystal Dew World: https://crystalmark.info/
----------------------------------------------------------------------------

    OS : Windows Server 2016 Server Standard (full installation) [10.0 Build 14393] (x64)
  Date : 2023/04/01 15:50:57

-- Controller Map ----------------------------------------------------------
 + Contrôleur IDE standard double canal PCI [ATA]
   - ATA Channel 0 (0)
   + ATA Channel 1 (1)
     - hp DVD-RAM GH80N ATA Device
 - Contrôleur des espaces de stockage Microsoft [SCSI]
 + Contrôleur Smart Array P410i (pilote de média) [SCSI]
   - N° LUN virtuel HP
   - HP LOGICAL VOLUME SCSI Disk Device

-- Disk List ---------------------------------------------------------------

there is the log 
**__________________________________________________**

and there is a hwinfo report 
Computer:      HP ProLiant ML350 G6
CPU:           Intel Xeon E5620 (Westmere-EP, B1)
               2400 MHz (18.00x133.3) @ 1599 MHz (12.00x133.3)
BIOS:          D22, 05/05/2011
Chipset:       Intel 5520 (Tylersburg 36D) + ICH10 Consumer
Memory:        12288 MBytes @ 533 MHz, 7-7-7-20
               - 8192 MB PC10600 DDR3 SDRAM - Samsung M393B1K70CH0-CH9
               - 4096 MB PC10600 DDR3 SDRAM - Samsung M393B5270CH0-CH9
Graphics:      AMD ES1000 (RN50) [Hewlett-Packard]
               AMD ES1000, 64 MB DDR2 SDRAM
Drive:         hp DVD-RAM GH80N, DVD+R DL
Drive:         COMPAQ SCSI COMMUNICATE, Unknown or none
Drive:         HP LOGICAL VOLUME, Disk drive
Network:       Broadcom BCM5714 NetXtreme Gigabit Ethernet Controller
Network:       Broadcom BCM5714 NetXtreme Gigabit Ethernet Controller
Network:       RealTek Semiconductor RTL8188CE Wireless LAN 802.11n PCI-E Network Adapter
Network:       Broadcom NetXtreme Gigabit Ethernet
OS:            Microsoft Windows Server 2016 Standard (x64) Build 14393.693 (1607/RS1)

Felzow47, 2023-04-01:
> Hi
> 
> And with the 9.0.0 Beta 4 ?
> 
> https://crystalmark.info/en/
> 
> Cheers !

same error with beta 9

hiyohiyo, 2023-04-03:
Unfortunately, the Smart Array P410i controller is not supported.

closed 2023-04-03 by hiyohiyo