hiyohiyo/CrystalDiskInfo #238: When the platform toolset is set to v140_xp, the symbol BusTypeNvme is not defined in the 64-bit program

AsunaU2, 2023-12-11:
When the platform toolset is set to v140_xp, the symbol BusTypeNvme is not defined in the 64-bit program:
[image]

At the same time, I noticed that under WIN64, BusTypeNvme is defined as 17. Can I directly set it to 17 to ensure that the compilation passes? Like this:
[image]

I don't know if doing this will affect functionality in some cases

Charltsing, 2023-12-19:
https://learn.microsoft.com/en-us/windows-hardware/drivers/ddi/ntddstor/ne-ntddstor-storage_bus_type

AsunaU2, 2023-12-20:
> https://learn.microsoft.com/en-us/windows-hardware/drivers/ddi/ntddstor/ne-ntddstor-storage_bus_type 你数一下，这玩意正好是第17个
> 
> https://learn.microsoft.com/zh-cn/windows/win32/api/VdsLun/ne-vdslun-vds_storage_bus_type

Yes, so I'm wondering if the original code was written to prevent future changes to the enumeration value

Charltsing, 2023-12-20:
> Yes, so I'm wondering if the original code was written to prevent future changes to the enumeration value

微软的定义你可以放心，不会改的