hiyohiyo/CrystalDiskInfo #239: warn when SMART (C7) Ultra DMA CRC Error Count and BB not correctable errors increase

nona00, 2024-01-06:
Hello,

Is it possible to enable a warning when SMART (C7) Ultra DMA CRC Error Count and BB not correctable error
increases? 

I had an issue with SATA cables/backplane and CrystalDiskInfo 9.2.1 x64 did not warn.

[image]

[image]

WyvernIXTL, 2024-02-15:
I'd like to know that as well!
I had an SSD die just like that with drive health 100%. BB (uncorrectable errors) growing every day, killing more and more of my personal files.

WyvernIXTL, 2024-02-15:
If this doesn't get implemented, I might just have to write a script with regex and  [`smartctl`](https://github.com/smartmontools/smartmontools). >_<

I don't feel like having to detect drive failure late again.

nona00, 2024-03-05:
Here another Samsung SSD which is defective, but no warning in CrystalDiskInfo:

[image]