hiyohiyo/CrystalDiskInfo #251: SanDisk SSD P4 regression starting with version 8.8.0.

RedGreenBlue09, 2024-06-14:
Before:
[image]

After:
[image]

Total host Reads: Looks like it's reading garbage memory as the number changes each instance.  
Total host Writes: Gone.
Health Status: 100% instead of 97%.

hiyohiyo, 2024-06-15:
I may fix CrystalDiskInfo 9.3.1.

RedGreenBlue09, 2024-06-15:
Hi, the Dashboard software only gives me raw values. Furthermore, serial number seems to be wrong too.
[image] [image]
[image] [image]

hiyohiyo, 2024-06-15:
Thank you for testing.
I will fix SanDisk SSD P4 support in CrystalDiskInfo 9.3.1.
I would like you to test latest version.
https://crystalmark.info/

RedGreenBlue09, 2024-06-15:
Thanks, the read & write values returned. What about the 97% health that version 8.7.0 gives me?
[image]

RedGreenBlue09, 2024-06-15:
I can notice the value `C7` has lost its description. I used Victoria HDD and GSmartControl, they both shows it as "Ultra DMA CRC errors`.

RedGreenBlue09, 2024-06-18:
To clarify, this drive is a mSATA SSD connected through a USB adaptor.