hiyohiyo/CrystalDiskInfo #277: How to convert a project into a dynamic link library (如何将项目转换为动态链接库)

zyyujq, 2025-05-13:
CrystalDiskInfo 8.17.14 can be successfully converted to DLL (CrystalDiskInfo 8.17.14 可以成功转换为DLL)

Just quote （只要引用 ）

AtaSmart.cpp
AtaSmart.h
DnpService.h
NVMeInterpreter.cpp
NVMeInterpreter.h
OsInfoFx.cpp
OsInfoFx.h
SlotSpeedGetter.cpp
SlotSpeedGetter.h
SPTIUtil.h
StorageQuery.h
UtilityFx.cpp
UtilityFx.h

Why CrystalDiskInfo 9.7 cannot be migrated to DLL (为什么 CrystalDiskInfo 9.7 无法迁移为 DLL)

zyyujq, 2025-05-13:
#pragma once

#include "AtaSmart.h"
#include "NVMeInterpreter.h"

extern "C" __declspec(dllexport) CAtaSmart* CreateAtaSmart()
{
	auto ata = new CAtaSmart;
	ata->SetAtaPassThroughSmart(TRUE);
	ata->FlagUsbSat = TRUE;
	ata->FlagUsbIodata = TRUE;
	ata->FlagUsbSunplus = TRUE;
	ata->FlagUsbLogitec = TRUE;
	ata->FlagUsbProlific = TRUE;
	ata->FlagUsbJmicron = TRUE;
	ata->FlagUsbCypress = TRUE;
	ata->FlagUsbASM1352R = TRUE;
	ata->FlagUsbMemory = TRUE;
	ata->FlagUsbNVMeJMicron = TRUE;
	ata->FlagUsbNVMeASMedia = TRUE;
	ata->FlagUsbNVMeRealtek = TRUE;
	ata->FlagMegaRAID = TRUE;
	return ata;
}

extern "C" __declspec(dllexport) void DestroyAtaSmart(CAtaSmart* ptr)
{
	if (ptr)
	{
		delete ptr;
		ptr = nullptr;
	}
}

extern "C" __declspec(dllexport) void InitAtaSmart(CAtaSmart* ptr, bool useWmi, bool advancedDiskSearch,
                                                   bool flagChangeDisk, bool workaroundHD204UI, bool workaroundAdataSsd,
                                                   bool flagHideNoSmartDisk)
{
	auto flagCD = flagChangeDisk ? TRUE : FALSE;
	ptr->Init(useWmi ? TRUE : FALSE, advancedDiskSearch ? TRUE : FALSE, &flagCD, workaroundHD204UI ? TRUE : FALSE,
	          workaroundAdataSsd ? TRUE : FALSE, flagHideNoSmartDisk ? TRUE : FALSE);
}

char* csToString(CString str)
{
	auto len = str.GetLength() + 1;
	auto buff = new TCHAR[len];
	memcpy(buff, (LPCTSTR)str, sizeof(TCHAR) * (len));
	auto charBuf = (char*)CoTaskMemAlloc(len);
	size_t converted;
	wcstombs_s(&converted, charBuf, len, buff, len);
	delete[] buff;
	return charBuf;
}

extern "C" __declspec(dllexport) int GetPhysicalDriveId(CAtaSmart* ptr, int index)
{
	return ptr->vars[index].PhysicalDriveId;
}

extern "C" __declspec(dllexport) int GetInterfaceType(CAtaSmart* ptr, int index)
{
	return ptr->vars[index].InterfaceType;
}

extern "C" __declspec(dllexport) char* GetModel(CAtaSmart* ptr, int index)
{
	return csToString(ptr->vars[index].Model);
}

//extern "C" __declspec(dllexport) char* GetMod(int index)
//{
//	CAtaSmart m_ATA;
//	m_ATA.Init(TRUE, FALSE, NULL, TRUE, TRUE, FALSE);
//	auto str= csToString(m_ATA.vars[index].Model);
//	return str;
//}

extern "C" __declspec(dllexport) char* GetSerialNumber(CAtaSmart* ptr, int index)
{
	return csToString(ptr->vars[index].SerialNumber);
}

//extern "C" __declspec(dllexport) char* GetSN(int index)
//{
//	CAtaSmart m_ATA;
//	m_ATA.Init(TRUE, FALSE, NULL, TRUE, TRUE, FALSE);
//	auto str = csToString(m_ATA.vars[index].SerialNumber);
//
//	return str;
//
//}

extern "C" __declspec(dllexport) char* GetFirmware(CAtaSmart* ptr, int index)
{
	return csToString(ptr->vars[index].FirmwareRev);
}

extern "C" __declspec(dllexport) _SMART_ATTRIBUTE* GetSmart(CAtaSmart* ptr, int index)
{
	return ptr->vars[index].Attribute;
}

extern "C" __declspec(dllexport) unsigned long GetTotalDiskSize(CAtaSmart* ptr, int index)
{
	return ptr->vars[index].TotalDiskSize;
}

extern "C" __declspec(dllexport) int GetDiskCount(CAtaSmart* ptr)
{
	return ptr->vars.GetCount();
}

extern "C" __declspec(dllexport) bool CreateRandomFile(const char* path, int size)
{
	CString TestFilePath(path);
	auto hFile = ::CreateFile(TestFilePath, GENERIC_READ | GENERIC_WRITE, 0, NULL, CREATE_ALWAYS,
	                          FILE_ATTRIBUTE_NORMAL | FILE_FLAG_NO_BUFFERING | FILE_FLAG_SEQUENTIAL_SCAN, NULL);

	if (hFile == INVALID_HANDLE_VALUE)
	{
		return false;
	}

	// COMPRESSION_FORMAT_NONE
	USHORT lpInBuffer = COMPRESSION_FORMAT_NONE;
	DWORD lpBytesReturned = 0;
	DeviceIoControl(hFile, FSCTL_SET_COMPRESSION, (LPVOID)&lpInBuffer,
	                sizeof(USHORT), NULL, 0, (LPDWORD)&lpBytesReturned, NULL);

	// Fill Test Data
	char* buf = NULL;
	int BufSize;
	int Loop;
	int i;
	DWORD writesize;
	BufSize = 1024 * 1024;
	Loop = size;

	buf = (char*)VirtualAlloc(NULL, BufSize, MEM_RESERVE | MEM_COMMIT, PAGE_READWRITE);
	if (buf == NULL)
	{
		return false;
	}

	for (i = 0; i < BufSize; i++)
	{
		buf[i] = (char)(rand() % 256);
	}

	for (i = 0; i < Loop; i++)
	{
		WriteFile(hFile, buf, BufSize, &writesize, NULL);
	}
	VirtualFree(buf, 0, MEM_RELEASE);
	CloseHandle(hFile);
	return true;
}

zyyujq, 2025-05-13:
CrystalDiskInfo 8.17.14 can be successfully converted to DLL (CrystalDiskInfo 8.17.14 可以成功转换为DLL)
Can it be modified and up

Can it be modified and updated to V9.7?  （可以修改完善更新到V9.7？）

zyyujq, 2025-05-13:
[Diskinfo.zip](https://github.com/user-attachments/files/20187572/Diskinfo.zip)

zyyujq, 2025-05-13:
[Diskinfo_compile.zip](https://github.com/user-attachments/files/20187641/Diskinfo_compile.zip)

Charltsing, 2025-05-13:
我是直接把代码翻译成c#，用着挺好的

zyyujq, 2025-05-14:
> 我是直接把代码翻译成c#，用着挺好的

共享一下

Charltsing, 2025-05-16:
刚抽取测试了你的dll，有两个问题
1、大部分硬盘不需要管理员权限就可以读取特征码。所以你可以修改一下CrystalDiskInfo的源代码，去掉管理员依赖
2、读取硬盘信息的速度太慢，不知道是不是保留了WMI造成的。正常情况下，通过API读取硬盘信息是[微秒](https://www.52pojie.cn/thread-1850119-1-1.html)级的。